pybluez/pybluez #79: Kivy

rossayoung, 2015-04-06:
Hi Karulis,  I really like the stuff you are doing with pybluez.  I was wondering if there was an easy way to import pybluez into kivy?  This would allow kivy apps bluetooth functionality without having to run the hcitool like commands thus enabling kivy apps to be ported on more Operating Systems.

karulis, 2015-04-07:
Well pybluez now supports windows stack and linux with bluez.

There is no support for ios and it will require lot of work.

Regarding android - as far as I know(thought I am no android expert) now android uses by default bludroid stack, and again it will require lot of work to support it in pybluez.
But if it is possible to to use bluez instead of bludroid then pybluez should work with no problems.

Support for ios and osx is in future plans, as is Bluetooth Low Energy.

karulis, 2015-05-14:
I will close that one and leave issue https://github.com/karulis/pybluez/issues/63 open

closed 2015-05-14 by karulis