pybluez/pybluez #81: How to reuse socket address in windows

aodock, 2015-05-13:
Hi,

I tried the example codes,.. they worked fine on linux. When i tried it on windows i got the following error saying,
```
server_sock.bind(("",port))
  File "C:\Python27\lib\site-packages\bluetooth\msbt.py", line 60, in bind
    status = bt.bind (self._sockfd, addr, port)
IOError: Only one usage of each socket address (protocol/network address/port) is normally permitted.
```
and i [found](http://stackoverflow.com/a/12362623/3869739) that, in `TCP/IP` there is a option to reuse the address. Is there a similar approach in pyblues to do so ?

karulis, 2015-05-13:
Could you point to or paste code that caused this exception?

To reuse the address w would need to implement setsockopt as mentioned in issue:
https://github.com/karulis/pybluez/issues/10

aodock, 2015-05-13:
i used [this](http://people.csail.mit.edu/albert/bluez-intro/x232.html) code and also [this one](http://blog.kevindoran.co/bluetooth-programming-with-python-3/),  first on linux, i,e. both server and client code on 2 different linux machine,.. it worked. When i tried server side code on windows i got that error.

And yes, i went through the the issue:#10,  and im sorry that i couldnt figure out anything. Can you plz point out an example if any.

karulis, 2015-05-13:
Try changing port number in both scripts in http://people.csail.mit.edu/albert/bluez-intro/x232.html
On my win box RFCOMM ports 1-7 are already taken by other services
Try for example 17 and let me know if it helps

karulis, 2015-05-14:
Closing, feel free to reopen.

closed 2015-05-14 by karulis