pybluez/pybluez #92: advertise_service() uncorrectly work in some condition.

KHU-YoungBo, 2015-08-18:
advertise_service() uncorrectly work in some condition.
 - Host computer has serveral bt devices.
 - The socket which we want to adversise is bound to specific device.
 - The device is un-advertisable.

A couple of days ago, I already mentioned this bug.

https://groups.google.com/forum/#!topic/pybluez/al8MDolLBd0

karulis, 2015-08-18:
Hi and thanks for report and great pull request:)

I just have one question - when the device is un-advertisable?

KHU-YoungBo, 2015-08-18:
- No device
- No up
- No running
- No iscan
- No pscan

I consider these state.

In addition, I found out these state can be read from flags in "struct hci_dev_info" using ioctl

karulis, 2015-08-20:
Thanks:)
I have tested it and it works fine:)
Will be included in release 0.23

karulis, 2015-08-28:
Merged

closed 2015-08-28 by karulis