pybluez/pybluez #96: python3.5 deprecated API warnings

karulis, 2015-09-22:
Lots of deprecated API warnings when building for python 3.5

msbt\_msbt.c(170): warning C4996: 'WSAStringToAddressA': Use WSAStringToAddressW() instead or define _WINSOCK_DEPRECATED
_NO_WARNINGS to disable deprecated API warnings
C:\Program Files (x86)\Windows Kits\8.1\include\um\winsock2.h(3623): note: see declaration of 'WSAStringToAddressA'
msbt\_msbt.c(241): warning C4996: 'WSAStringToAddressA': Use WSAStringToAddressW() instead or define _WINSOCK_DEPRECATED
_NO_WARNINGS to disable deprecated API warnings
C:\Program Files (x86)\Windows Kits\8.1\include\um\winsock2.h(3623): note: see declaration of 'WSAStringToAddressA'
msbt\_msbt.c(358): warning C4996: 'WSADuplicateSocketA': Use WSADuplicateSocketW() instead or define _WINSOCK_DEPRECATED
_NO_WARNINGS to disable deprecated API warnings
C:\Program Files (x86)\Windows Kits\8.1\include\um\winsock2.h(2872): note: see declaration of 'WSADuplicateSocketA'
msbt\_msbt.c(365): warning C4996: 'WSASocketA': Use WSASocketW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to di
sable deprecated API warnings
C:\Program Files (x86)\Windows Kits\8.1\include\um\winsock2.h(3457): note: see declaration of 'WSASocketA'
msbt\_msbt.c(479): warning C4996: 'WSAStringToAddressA': Use WSAStringToAddressW() instead or define _WINSOCK_DEPRECATED
_NO_WARNINGS to disable deprecated API warnings
C:\Program Files (x86)\Windows Kits\8.1\include\um\winsock2.h(3623): note: see declaration of 'WSAStringToAddressA'
msbt\_msbt.c(585): warning C4996: 'WSALookupServiceBeginA': Use WSALookupServiceBeginW() instead or define _WINSOCK_DEPR
ECATED_NO_WARNINGS to disable deprecated API warnings
C:\Program Files (x86)\Windows Kits\8.1\include\um\winsock2.h(3693): note: see declaration of 'WSALookupServiceBeginA'
msbt\_msbt.c(607): warning C4996: 'WSALookupServiceNextA': Use WSALookupServiceNextW() instead or define _WINSOCK_DEPREC
ATED_NO_WARNINGS to disable deprecated API warnings
C:\Program Files (x86)\Windows Kits\8.1\include\um\winsock2.h(3752): note: see declaration of 'WSALookupServiceNextA'
msbt\_msbt.c(720): warning C4996: 'WSASetServiceA': Use WSASetServiceW() instead or define _WINSOCK_DEPRECATED_NO_WARNIN
GS to disable deprecated API warnings
C:\Program Files (x86)\Windows Kits\8.1\include\um\winsock2.h(4157): note: see declaration of 'WSASetServiceA'
msbt\_msbt.c(775): warning C4996: 'WSASetServiceA': Use WSASetServiceW() instead or define _WINSOCK_DEPRECATED_NO_WARNIN
GS to disable deprecated API warnings
C:\Program Files (x86)\Windows Kits\8.1\include\um\winsock2.h(4157): note: see declaration of 'WSASetServiceA'

beadysea, 2019-07-28:
I get the same warnings on Windows 10 with Python 3.6 and 3.7. Seems to be changes in winsock2 between windows 7 and windows 8.1 releases. 
[Compatibility Win7.0 to Win8.1](https://abi-laboratory.pro/compatibility/Windows_7.0_to_Windows_8.1/x86_64/headers_diff/ws2_32.dll/diff.html)
Looking around the net, the easy is to just ignore the compiler warnings by adding
 `#define _WINSOCK_DEPRECATED_NO_WARNINGS`
before `#include <winsock2.h>` in _msbt.c
A simple fix however if we do this I think it's important to comment the offending lines with todo's. Compilation warnings are there for a reason after all.

rgov, 2019-07-30:
There is a branch, https://github.com/pybluez/pybluez/tree/issue_96, where this was worked on previously. You can see the diff here: https://github.com/pybluez/pybluez/compare/issue_96?expand=1

I'm not sure what the status is.

PR #269 (2019-10-16): Fix deprecated API warnings, close #96
This patch resolves #96.
I tested this on Windows 10 using python 3.6 32 bit and python 3.7 64 bit by installing from source and running the example inquiry.py. I don't have access to earlier versions of Windows so can't test backwards compatibility.
 
I've attached the egg file for the community to test on earlier windows versions. 
Download [PyBluez-0.22-py3.6-win32.zip](https://github.com/pybluez/pybluez/files/3508727/PyBluez-0.22-py3.6-win32.zip) and rename to PyBluez-0.22-py3.6-win32.egg
commit 1fb65b9a (2019-10-16): Merge pull request #269 from beadysea/issue96-patch

Fix deprecated API warnings, close #96
closed 2019-10-16 by rgov