pybluez/pybluez #97: Docs are difficult to view

NelsonCrosby, 2015-09-28:
I currently don't see a way to view the docs other than cloning. This isn't optimal. There should probably also be some hint on how to view them.

- The quickest fix would probably be to pull the existing docs into `gh-pages`, but it may be more difficult if it is desired to keep the docs in `master` as well.
- If someone feels like rewriting the docs in markdown, it's easy enough to view them directly on GitHub, and via http://viewdocs.io.
- If someone feels like rewriting the docs in markdown or reStructuredText, they can be hosted on https://readthedocs.org.

NelsonCrosby, 2015-10-13:
I'll take a look at rewriting these when I next get some spare time. Unless there's a stated preference towards RST, I'll use Markdown. Might be a few weeks before I get started, though.

karulis, 2015-11-01:
Any help will be more then appreciated :)
The only suggestion is that it would be nice if we could also place documentation on pythonhosted.org

NelsonCrosby, 2015-11-07:
That (hopefully) shouldn't be too difficult to do. Might just need to add something of a script to populate an HTML template so it doesn't look ugly.

I'm going to get started doing MD docs now. How are the docs currently being created? Are they generated, and should I re-generate them before I start?

NelsonCrosby, 2015-11-07:
I've started and have got [a checklist](https://github.com/NelsonCrosby/pybluez/blob/master/docs-rewrite/docs-todo.md) in place.

alexandroid000, 2016-03-24:
You can use Pandoc to convert the HTML to markdown and it would save you some work, if you haven't gotten too far yet

NelsonCrosby, 2016-03-25:
I never ended up getting around to it, and I've ended up busy as hell again. Sorry, really wish I could have done something.

In terms of HTML-to-<w/e> for the existing docs, the current HTML is done using iframes or something, so I don't know how well other tools will work with that. Can't remember if I ended up working out the layout or anything, either.

IIRC, the actual docs weren't even close to comprehensive, and I couldn't work out how they were generated. I did some research into generating something from docstrings, but the furthest I got was tools to generate HTML (which ends up not helping us in any way).

fenwick67, 2016-11-23:
In a pinch, rawgit will deliver the HTML and assets: https://cdn.rawgit.com/karulis/pybluez/master/docs/index.html

beadysea, 2019-07-11:
I'd like to contribute to this project in some way and I think documentation could be a good start. I'm thinking that given this issue is old it could be worth the time to rebuild the documentation from the ground up. I like @NelsonCrosby's idea of using reStructuredText so they can be hosted on https://readthedocs.org and/or https://pythonhosted.org. 

I've used github for my own projects but never contributed to another project. I've forked this repo and propose to create an issue-97 branch on my fork which I would then pull request on this repo. Would that be the right way to progress @karulis? I'm also considering using sphynx to help automate the process and broadly following [@NelsonCrosby's checklist.](https://github.com/NelsonCrosby/pybluez/blob/master/docs-rewrite/docs-todo.md)

Look forward to hearing views from the community.

rgov, 2019-07-11:
@beadysea We would love your contribution. Forking and creating a pull request would be a good way to go about it. Let me know if I can help.

PR #244 (2019-07-23): Generate documentation using Sphinx
Starting work to resolve #97 with improved documentation. So far I've started the process of writing restructured text files containing Sphinx directives. By using sphinx we can publish the projects documentation on [readthedocs.org
](https://readthedocs.org/)

Docs so far:
api.rst - (API Docs) Developers interface with autogenerated documentation from source code docstrings.
index.rst - Project homepage 

Checklist:
- [x] Work on API docs
    - [ ] Update source code doctrings.
- [x] Table of Contents
- [x] Replace old docs with new docs
- [ ] Create user d [... cut]
closed 2019-07-23 by rgov