pybluez/pybluez #99: Bluetooth module not working

sudiptakarmakar, 2015-10-13:
When I'm running this code
```
import bluetooth as bt
bt.discover_devices(lookup_names=True)
```
I get the following error:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/piyush/.virtualenvs/py27/local/lib/python2.7/site-packages/bluetooth/bluez.py", line 26, in discover_devices
    device_id = _bt.hci_get_route()
OSError: [Errno 19] No such device
```

Also, when I'm calling `advertise_service` following the examples provided, I got the following errors in different times,
```
Traceback (most recent call last):
  File "build/bdist.linux-x86_64/egg/bluetooth/bluez.py", line 268, in advertise_service
  bluetooth.btcommon.BluetoothError: error no advertisable device.
```

```
Traceback (most recent call last):
  File "bt.py", line 17, in <module>
    advertise_service( server_sock, "SampleServer", service_id = uuid, service_classes = [ uuid, SERIAL_PORT_CLASS ], profiles = [ SERIAL_PORT_PROFILE ])
  File "/home/machinename/.virtualenvs/py27/local/lib/python2.7/site-packages/bluetooth/bluez.py", line 242, in advertise_service
    raise BluetoothError (str (e))
bluetooth.btcommon.BluetoothError: (2, 'No such file or directory')
```
Here is my [stackoverflow post](http://stackoverflow.com/questions/33110992/python-code-for-bluetooth-throws-error-after-i-had-to-reset-the-adapter) about it.

karulis, 2015-11-01:
So according to stack overflow I can close that one

closed 2015-11-01 by karulis

poooooooooooooi, 2017-11-04:
metoo