pybluez/pybluez #102: Rfcomm socket drops connection on OSX

karulis, 2015-11-01:
When running example scripts: rfcomm-client.py/ rfcomm-server.py on OSX 
rfcomm socket closes connection

cspensky, 2016-05-18:
I'm having this same problem even when running a native Obj-C program on OSX.  Seems like something must have changed in their API, yet I'm struggling to find documentation for it.

karulis, 2016-05-21:
With pyobjc 3.1.1 when I try to run client on OSX I get:

> Searching for SampleServer on 28:E3:47:BB:2F:84
> Traceback (most recent call last):
>   File "rfcomm-client.py", line 25, in <module>
>     service_matches = find_service( uuid = uuid, address = addr )
>   File "build/bdist.macosx-10.11-intel/egg/bluetooth/osx.py", line 49, in find_service
>   File "build/bdist.macosx-10.11-intel/egg/lightblue/_lightblue.py", line 103, in findservices
>   File "build/bdist.macosx-10.11-intel/egg/lightblue/_lightblue.py", line 529, in _getservicetuple
> TypeError: Need 1 arguments, got 0
>

cspensky, 2016-05-27:
I think I fixed the problem that you're having.  They changed some things in pyobjc.  Should I submit a pull request?

cspensky, 2016-05-27:
It looks like the connection is getting dropped due to OSX's attempt to automatically connect to all services.

**[May 27 15:33:24.129]  [NOTE]  Attempt to open L2CAP Channel with PSM 25 from process "bluetoothaudiod (24811)"**
[May 27 15:33:24.131]  [L2CAP SEND]  Connection Request - AVDTP
[May 27 15:33:24.136]  [L2CAP RECEIVE]  Connection Response - AVDTP - Connection Pending
[May 27 15:33:24.138]  [L2CAP RECEIVE]  Connection Response - AVDTP - Connection refused - security block
**[May 27 15:33:24.722]  [NOTE]  Attempt to close L2CAP Channel 0x0041 from process "bluetoothaudiod (24811)"**
[May 27 15:33:24.722]  [RFCOMM SEND]  DISC for channel 1 
[May 27 15:33:24.722]  [RFCOMM SEND]  DISC for channel 0 
[May 27 15:33:24.728]  [RFCOMM RECEIVE]  UA for channel 1 
[May 27 15:33:24.729]  [RFCOMM RECEIVE]  UA for channel 0 
[May 27 15:33:24.729]  [L2CAP SEND]  Disconnection Request - RFCOMM
[May 27 15:33:24.731]  [L2CAP RECEIVE]  Disconnection Request - RFCOMM
[May 27 15:33:24.731]  [L2CAP SEND]  Disconnection Response - RFCOMM
[May 27 15:33:24.731]  [HCI COMMAND]  [0406] Disconnect - Connection Handle: 0x000B - Reason: Other End Terminated Connection: User Ended Connection

karulis, 2016-05-28:
> I think I fixed the problem that you're having. They changed some things in pyobjc. Should I submit a pull request? 
@cspensky off course if You have fix then I will be happy to merge it:)

closed 2019-07-30 by stale[bot]