pybluez/pybluez #103: Running inquiry.py on OSX causes segmentation fault

karulis, 2015-11-01:
I will attach log

karulis, 2015-11-01:
Process:               Python [4782]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            org.python.python
Version:               2.7.10 (2.7.10)
Build Info:            python-89040001000000~1
Code Type:             X86-64 (Native)
Parent Process:        bash [4694]
User ID:               501

PlugIn Path:             /Library/Python/2.7/site-packages/pyobjc_core-3.1b1-py2.7-macosx-10.10-intel.egg/objc/_objc.so
PlugIn Identifier:       _objc.so
PlugIn Version:          ??? (0)

Date/Time:             2015-10-15 16:21:56.676 +0200
OS Version:            Mac OS X 10.10.5 (14F27)
Report Version:        11
Anonymous UUID:        51F16D7D-47E6-2008-2666-315CBEC55A5E

Sleep/Wake UUID:       41746073-445F-40C3-AEB2-3065EDBDBEDB

Time Awake Since Boot: 250000 seconds
Time Since Wake:       140 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000001

VM Regions Near 0x1:
--> 
    __TEXT                 0000000106446000-0000000106447000 [    4K] r-x/rwx SM=COW  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.python.python             	0x000000010647830a PyList_Append + 145
1   _objc.so                      	0x00000001068339c0 method_stub + 967 (libffi_support.m:1261)
2   libffi.dylib                  	0x00007fff8e34ea07 ffi_closure_unix64_inner + 511
3   libffi.dylib                  	0x00007fff8e34e0c6 ffi_closure_unix64 + 70
4   com.apple.Bluetooth           	0x00007fff8a62dcf0 -[IOBluetoothDeviceInquiry deviceInquiryUpdatingDeviceNamesStarted:] + 97
5   com.apple.CoreFoundation      	0x00007fff947fc7bc __invoking___ + 140
6   com.apple.CoreFoundation      	0x00007fff947fc612 -[NSInvocation invoke] + 290
7   com.apple.CoreFoundation      	0x00007fff9489c5c6 -[NSInvocation invokeWithTarget:] + 54
8   com.apple.CoreFoundation      	0x00007fff9486dc26 ___forwarding___ + 518
9   com.apple.CoreFoundation      	0x00007fff9486d998 _CF_forwarding_prep_0 + 120
10  com.apple.CoreFoundation      	0x00007fff947fc7bc __invoking___ + 140
11  com.apple.CoreFoundation      	0x00007fff947fc612 -[NSInvocation invoke] + 290
12  com.apple.Foundation          	0x00007fff936ceaed -[NSConnection dispatchInvocation:] + 138
13  com.apple.Foundation          	0x00007fff936cdbad -[NSConnection handleRequest:sequence:] + 1360
14  com.apple.Foundation          	0x00007fff936be075 -[NSConnection handlePortCoder:] + 657
15  com.apple.Foundation          	0x00007fff936bd9b9 -[NSConnection dispatchWithComponents:] + 50
16  com.apple.Foundation          	0x00007fff936a2dea __NSFireMachPort + 254
17  com.apple.CoreFoundation      	0x00007fff94841dad __CFMachPortPerform + 285
18  com.apple.CoreFoundation      	0x00007fff94841c79 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__ + 41
19  com.apple.CoreFoundation      	0x00007fff94841beb __CFRunLoopDoSource1 + 475
20  com.apple.CoreFoundation      	0x00007fff94833767 __CFRunLoopRun + 2375
21  com.apple.CoreFoundation      	0x00007fff94832bd8 CFRunLoopRunSpecific + 296
22  com.apple.HIToolbox           	0x00007fff8a7af56f RunCurrentEventLoopInMode + 235
23  com.apple.HIToolbox           	0x00007fff8a7af2ea ReceiveNextEventCommon + 431
24  com.apple.HIToolbox           	0x00007fff8a7af12b _BlockUntilNextEventMatchingListInModeWithFilter + 71
25  com.apple.AppKit              	0x00007fff8cc398ab _DPSNextEvent + 978
26  com.apple.AppKit              	0x00007fff8cc38e58 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 346
27  libffi.dylib                  	0x00007fff8e34df44 ffi_call_unix64 + 76
28  libffi.dylib                  	0x00007fff8e34e781 ffi_call + 853
29  _objc.so                      	0x000000010683924b PyObjCFFI_Caller + 2064 (libffi_support.m:4032)
30  _objc.so                      	0x0000000106857bea objcsel_call + 163 (selector.m:505)
31  org.python.python             	0x000000010645750a PyObject_Call + 99
32  org.python.python             	0x00000001064d4399 PyEval_EvalFrameEx + 11795
33  org.python.python             	0x00000001064d1352 PyEval_EvalCodeEx + 1409
34  org.python.python             	0x00000001064d7bf1 _PyEval_SliceIndex + 757
35  org.python.python             	0x00000001064d49a6 PyEval_EvalFrameEx + 13344
36  org.python.python             	0x00000001064d1352 PyEval_EvalCodeEx + 1409
[... 298 lines cut ...]
__LINKEDIT                         71.6M
__TEXT                            109.4M
__UNICODE                           552K
mapped file                        43.6M
shared memory                         4K
===========                      =======
TOTAL                             395.9M

Model: Macmini7,1, BootROM MM71.0220.B03, 2 processors, Intel Core i5, 1.4 GHz, 4 GB, SMC 2.24f32
Graphics: Intel HD Graphics 5000, Intel HD Graphics 5000, Built-In
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80CE, 0x4B3445384533303445452D45474345000000
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80CE, 0x4B3445384533303445452D45474345000000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x13B), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.6f3 16238, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD HTS545050A7E362, 500,11 GB
USB Device: USB2.0 Hub
USB Device: USB Receiver
USB Device: Composite Device
USB Device: USB2.0 Hub
USB Device: USB Optical Mouse
USB Device: IR Receiver
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: Mac mini, Apple Inc., 26.1

cspensky, 2016-05-18:
I've seen this as well, but am having trouble replicating it reliably.  Let me know if you have any advice for replicating it, and I'd happily help track it down.

karulis, 2016-05-21:
I have updated OSX to 10.11.4 and pyobjc to 3.1.1
and the issue seems to be solved
Let me know if it still happens for You

mdiviesti, 2017-09-20:
I'm on OSX 10.12.6
Python 3.6.2 (installed with brew)
I downloaded PyBlueZ master then changed setup.py to use python3. 
ran setup.py. 

pyobjc-core 3.2.1

Running the sample with this line: 
nearby_devices = bluetooth.discover_devices(duration=8, lookup_names=True, flush_cache=True, lookup_class=False)

I get a segment fault as described by the original ticket. 
If I remove lookup_class=False argument, it runs fine and discovers my devices (every single time without issue). Which is weird considering that lookup_class default is false in the method to begin with. I'm not sure why this would be an issue but it is.

closed 2019-07-30 by stale[bot]

Doophie, 2020-03-10:
Im having this issue now

rgov, 2020-03-10:
@Doophie This issue was last updated 2.5 years ago, please file a new issue.