pybluez/pybluez #108: Import errors

ericmjl, 2015-12-11:
I am experiencing import errors:

    In [1]: import bluetooth
    ---------------------------------------------------------------------------
    ImportError                               Traceback (most recent call last)
    <ipython-input-1-3a55efdd484d> in <module>()
    ----> 1 import bluetooth

    /Users/ericmjl/anaconda/lib/python3.4/site-packages/PyBluez-0.22-py3.4.egg/bluetooth/__init__.py in <module>()
         45         from bluetooth.bluez import *
         46 elif sys.platform == "darwin":
    ---> 47     from .osx import *
         48 else:
         49     raise Exception("This platform (%s) is currently not supported by pybluez." % sys.platform)

    /Users/ericmjl/anaconda/lib/python3.4/site-packages/PyBluez-0.22-py3.4.egg/bluetooth/osx.py in <module>()
    ----> 1 import lightblue
          2 from .btcommon import *
          3 
          4 def discover_devices(duration=8, flush_cache=True, lookup_names=False,
          5         lookup_class=False, device_id=-1):

    /Users/ericmjl/anaconda/lib/python3.4/site-packages/lightblue/__init__.py in <module>()
        158 
        159 # import implementation modules
    --> 160 from _lightblue import *
        161 from _lightbluecommon import *
        162 import obex     # plus submodule

    ImportError: No module named '_lightblue'

    In [2]: 

Not sure why this is the case. 

System is:

- Mac OS X El Capitan 10.11.1
- Anaconda Python.
- Modified `pybluez`'s `setup.py` script such that line 100 reads: 

    build_str = "sudo xcodebuild install -arch '%s' -target LightAquaBlue -configuration Release DSTROOT=/ INSTALL_PATH=/Library/Frameworks DEPLOYMENT_LOCATION=YES" % (arch)

- Above modification was done so that I could install LightAquaBlue.

Checking my Anaconda Python directory, I have the following items listed under the `lightblue` dir:

    In [26]: cd site-packages/lightblue/
    /Users/ericmjl/anaconda/lib/python3.4/site-packages/lightblue

    In [27]: ls
    _IOBluetooth.py       _LightAquaBlue.py     __pycache__/          _lightblue.py         _macutil.py           _obexcommon.py
    _IOBluetoothUI.py     __init__.py           _bluetoothsockets.py  _lightbluecommon.py   _obex.py              obex.py

Not sure what is causing these import errors - any ideas, by any chance?

karulis, 2015-12-11:
Just to rule one thing out - have You tried restarting system after installation?

ericmjl, 2015-12-11:
Yes, I have, and the same errors still come in. Might there be something I did incorrectly?

karulis, 2015-12-11:
Ok I believe I have reproduced it.
OSX support is still very unstable - but I thought installation is working.
Tomorrow I will have some time to fix it.
Thanks for reporting:)

karulis, 2015-12-12:
Ok its fixed, try latest repo version.
Let me know if it works:)

thomasleveil, 2015-12-12:
see 42070d95b

ericmjl, 2015-12-14:
It looks like it (mostly) worked! :smile: Thank you for taking the time to put up the fix :). 

More details on where some errors still show up are below:

(1) I still had to modify the `setup.py` script to use: `build_str = "sudo xcodebuild install -arch '%s' -target LightAquaBlue -configuration Release DSTROOT=/ INSTALL_PATH=/Library/Frameworks DEPLOYMENT_LOCATION=YES" % (arch)`. I think this may have something to do with El Capitan's modified permissions, in that we can't write to /Library without superuser permissions. I may be wrong, or blabbering incorrectly here! Please correct me if that's the case.

(2) On first import, I get the following error: 

    In [1]: import bluetooth
    /Users/ericmjl/anaconda/lib/python3.4/site-packages/objc/_bridgesupport.py:643: RuntimeWarning: Error parsing BridgeSupport data for IOBluetooth: PyObjCRT_SkipTypeSpec: Got '0x29' at end of union encoding, expecting '0x29'
      warnings.warn("Error parsing BridgeSupport data for %s: %s" % (frameworkName, e), RuntimeWarning)

That said, even with this, I can still use the API with no issues.

For fun, I tried doing a clean install in a new conda environment. Exact command for the conda environment was: `conda create -n "pybluez-clean" python=3.4"`. Using the same modifications as listed above, I get the following error:

    (pybluez-clean)Erics-MacBook-Air:~ ericmjl$ python
    Python 3.4.3 |Continuum Analytics, Inc.| (default, Oct 20 2015, 14:27:51) 
    [GCC 4.2.1 (Apple Inc. build 5577)] on darwin
    Type "help", "copyright", "credits" or "license" for more information.
    >>> import bluetooth
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "/Users/ericmjl/anaconda/envs/pybluez-clean/lib/python3.4/site-packages/PyBluez-0.22-py3.4.egg/bluetooth/__init__.py", line 47, in <module>
      File "/Users/ericmjl/anaconda/envs/pybluez-clean/lib/python3.4/site-packages/PyBluez-0.22-py3.4.egg/bluetooth/osx.py", line 1, in <module>
      File "/Users/ericmjl/anaconda/envs/pybluez-clean/lib/python3.4/site-packages/PyBluez-0.22-py3.4.egg/lightblue/__init__.py", line 160, in <module>
      File "/Users/ericmjl/anaconda/envs/pybluez-clean/lib/python3.4/site-packages/PyBluez-0.22-py3.4.egg/lightblue/_lightblue.py", line 23, in <module>
    ImportError: No module named 'Foundation'

Btw, I'm also at MIT (PhD in Course 20), happy to be your designated OSX software tester if it's still relevant! I'm hacking together an array of raspberry pis + bluetooth beacons to try simple proximity tracking on wild seals.

karulis, 2015-12-14:
> I still had to modify the setup.py script to use:  ... 
> can't write to /Library without superuser permissions. I may be wrong, or blabbering incorrectly here! Please correct me if that's the case.
Well I believe I had to run this command with sudo perms before - but I used it like "sudo python setup.py install"

> (2) On first import, I get the following error: ...
Second issue is probably related to pyobjc not yet properly handling El Capitan, and I like this runtime warning:)
> Got '0x29' at end of union encoding, expecting '0x29'

> ImportError: No module named 'Foundation'
I dont know exactly what - conda create -n "pybluez-clean" python=3.4" - is doing, but the error is related to missing pyobjc installation

>Btw, I'm also at MIT (PhD in Course 20), happy to be your designated OSX software tester if it's still relevant! I'm hacking together an array of raspberry pis + bluetooth beacons to try simple proximity tracking on wild seals. 
Maybe You could contribute to Pybluez BLE support on OSX?:)

ericmjl, 2015-12-14:
Yes, happy to help in small ways here and there!

karulis, 2015-12-15:
And BTW:
> That said, even with this, I can still use the API with no issues.

So You had no problems with issue #102 and #103 ?

ericmjl, 2015-12-15:
Those i didn't get to try, the only API calls I made were on finding devices.

karulis, 2015-12-16:
Thanks for information:)
Let me know if encounter any issues.

closed 2015-12-16 by karulis

codeundercoverdev, 2017-07-28:
Has anyone figured this out?  I have been working for hours to try and fix this but I keep getting the `ImportError: No module named '_lightblue'` error.

DeanMW, 2020-01-11:
Same issue here... No update?

rgov, 2020-01-11:
@DeanMW It's been 4 years since the initial issue was reported. Since then there have been many changes. Please make sure you are installing the latest version, 0.23, and file a new bug with more information that can be used to diagnose your issue.

anniethorburn, 2021-02-01:
I've spent a while banging my head against the `ImportError: No module named '_lightblue'` error and I've finally figured it out. It seems that many of the import statement in the source code for the lightblue package are incorrectly formatted. To fix it, go through the `.py` files in the `lightblue` dir and make the following changes:

- Wherever there is a `from X import *` statement in which X is file in the `lightblue` dir, change it to `from .X import *`. (E.g., change `from _lightblue import *` to `from ._lightblue import *`)
- Wherever there is an `import X` statement in which X is a file in the `lightblue` dir, change it to `from . import X`. (E.g., change `import _lightblue` to `from . import _lightblue`)

I hope this is helpful for someone!

akaler, 2021-03-19:
@anniethorburn can confirm this works