pybluez/pybluez #110: from bluetooth.ble import DiscoveryService causes GTK to crash

mstekker, 2016-01-24:
Hi,

After adding:
from bluetooth.ble import DiscoveryService

my GTK / Glade application crashes. This is happening during the initialization at a point where nothing calls the DiscoveryService. The GUI does not show up anymore.  The following is also imported:

import sys
import datetime
import os
import subprocess
from gi.repository import Gtk, GLib, GObject
import ConfigParser
from bluetooth.ble import DiscoveryService

Without GTK (just console) all is working fine. Any suggestions what could be the cause?

mstekker, 2016-01-31:
Update:

gattlib causes this issue. workaround: let the pygtk program call another python script (subprocess) which runs the discovery. This second script may not import any pygtk related modules. Not the best solutions - but it woks.

karulis, 2016-03-19:
I will close the issue and open one in gattlib. Thanks for reporting

closed 2016-03-19 by karulis