pybluez/pybluez #117: pybluez RFCOM detect connection loss

php-freak, 2016-03-15:
Is it possible to detect a connection loss with pybluez when using the RFCOM protocol?

egorf, 2016-03-18:
Pretty sure this is explained in the[ rfcomm server example](https://github.com/karulis/pybluez/blob/master/examples/simple/rfcomm-server.py):

```
print("Waiting for connection on RFCOMM channel %d" % port)

client_sock, client_info = server_sock.accept()
print("Accepted connection from ", client_info)

try:
    while True:
        data = client_sock.recv(1024)
        if len(data) == 0: break
        print("received [%s]" % data)
except IOError:
    pass

print("disconnected")

client_sock.close()
server_sock.close()
print("all done")
```

karulis, 2016-03-19:
I believe egorf comment says all. Closing

closed 2016-03-19 by karulis