pybluez/pybluez #118: Pairing and Connecting

egorf, 2016-03-18:
Hi!

This is more of a question, than an issue.
PyBluez is a great package to implement RFCOMM communication. However, I can't see why pairing bluetooth devices can not be implemented in the library. Are the only ways available on Linux are `bluetoothctl` utility and `simple-agent` like scripts?

karulis, 2016-03-19:
Well it should be possible to add that functionality not only for Linux,
but also for OSX and Windows.

bluetoothctl implementation uses dbus:
http://git.kernel.org/cgit/bluetooth/bluez.git/tree/client/main.c#n1209
there is also example in python:
http://git.kernel.org/cgit/bluetooth/bluez.git/tree/test/simple-agent

If You would like to implement this functionality for Linux based on dbus api,
then I would be happy to help You and merge it later

egorf, 2016-03-21:
Thanks for the tip!

closed 2016-03-21 by egorf