pybluez/pybluez #120: interrupt accept call

egorf, 2016-03-18:
Hi!

Is there a way to interrupt the blocking accept call in the RFCOMM server code?

Say, I run this an a separate process:
`client_socket, client_info = server_socket.accept()`

I end up waiting for a connection, before I can go any further. Is there a good way to stop waiting for the connection?

karulis, 2016-03-19:
It seems to be possible.
First You need to listen for incoming connections and then use select, some examples in C
http://stackoverflow.com/questions/3444729/using-accept-and-select-at-the-same-time
http://www.lowtek.com/sockets/select.html

egorf, 2016-03-21:
This is not possible via PyBluez then?

[There does not seem to be](https://pybluez.googlecode.com/svn/www/docs-0.7/public/bluetooth.BluetoothSocket-class.html#listen) a `select()` or `poll()` method.

karulis, 2016-03-21:
There is a select in python:
https://docs.python.org/3.5/library/select.html
it should work fine with pybluez

closed 2016-05-05 by karulis

muelli, 2017-08-19:
> Say, I run this an a separate process:

You mean a separate thread?

> Is there a way to interrupt the blocking accept call in the RFCOMM server code?

Shouldn't close()ing the socket (from a different thread) make the accept() return?