pybluez/pybluez #122: There is an IOError but the message is messy

yjcn, 2016-03-31:
Traceback (most recent call last):
  File "E:/python/Btgui/Gui.py", line 31, in run
    nearby_devices = bluetooth.discover_devices(duration=8, lookup_names=True, flush_cache=True, lookup_class=False)
  File "C:\Python27\lib\site-packages\bluetooth\msbt.py", line 15, in discover_devices
    devices = bt.discover_devices(duration=duration, flush_cache=flush_cache)
IOError: û�п��õ������ˡ�

My OS is Windows 10 Pro Simplified Chinese
And my code is using utf-8

karulis, 2016-05-05:
This log is copied directly from command line(cmd) window?
Can You Try to log it into file and check weather it is still messy?

yjcn, 2016-05-10:
I run it in cmd and the error message is Chinese "没有可用的数据了 " (GBK) It means there is no useful data.I want to know why the error happens and how fix it.
Mycode (QtCore.QThread):
    def run(self):
        print 'start thread'
        while not self.thread_stop:
            print 'scanning'
            try:
                nearby_devices = bluetooth.discover_devices(
                    duration=8, lookup_names=True, flush_cache=True, lookup_class=False)
            except IOError,e:
                print 'Error',e
            finally:
                for addr, name in nearby_devices:
                    self.emit(QtCore.SIGNAL('send'),addr,name)

karulis, 2016-05-15:
Have You tried running inquiry.py?

yjcn, 2016-05-18:
Yes. And I know why the error happens.When there is no device, the IOError happens.

karulis, 2016-05-21:
Well this is error reported by OS.
I dont think we can do much about it

yjcn, 2016-05-21:
Well.Thanks for your reply. : )

closed 2016-05-25 by karulis