pybluez/pybluez #133: Mac PyBluez installation failure: pyobjc 3.1.1, OS X 10.11.5

mlyons-tcc, 2016-07-06:
After installing PyObjc 3.1.1, I get an error when installing pybluez via `pip install pybluez`

```
    gcc -fno-strict-aliasing -I/Users/640labs/anaconda/envs/pybluez/include -arch x86_64 -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I/System/Library/Frameworks/IOBluetooth.framework/Headers -I/System/Library/Frameworks/CoreFoundation.framework/Headers -I/Users/640labs/anaconda/envs/pybluez/include/python2.7 -c osx/_osxbt.c -o build/temp.macosx-10.6-x86_64-2.7/osx/_osxbt.o
    osx/_osxbt.c:676:5: error: unknown type name 'IOBluetoothDeviceInquiryRef'; did you mean 'IOBluetoothDeviceRef'?
        IOBluetoothDeviceInquiryRef inquiry;
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
        IOBluetoothDeviceRef
    /System/Library/Frameworks/IOBluetooth.framework/Headers/IOBluetoothUserLib.h:34:47: note: 'IOBluetoothDeviceRef' declared here
    typedef struct OpaqueIOBluetoothObjectRef *                     IOBluetoothDeviceRef;
                                                                    ^
    osx/_osxbt.c:688:17: error: unknown type name 'IOBluetoothDeviceInquiryRef'; did you mean 'IOBluetoothDeviceRef'?
                    IOBluetoothDeviceInquiryRef inquiryRef,
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                    IOBluetoothDeviceRef
    /System/Library/Frameworks/IOBluetooth.framework/Headers/IOBluetoothUserLib.h:34:47: note: 'IOBluetoothDeviceRef' declared here
    typedef struct OpaqueIOBluetoothObjectRef *                     IOBluetoothDeviceRef;
                                                                    ^
    osx/_osxbt.c:703:19: warning: implicit declaration of function 'IOBluetoothDeviceInquiryCreateWithCallbackRefCon' is invalid in C99 [-Wimplicit-function-declaration]
        dd->inquiry = IOBluetoothDeviceInquiryCreateWithCallbackRefCon (&dd);
                      ^
    osx/_osxbt.c:703:17: warning: incompatible integer to pointer conversion assigning to 'IOBluetoothDeviceRef' (aka 'struct OpaqueIOBluetoothObjectRef *') from 'int' [-Wint-conversion]
        dd->inquiry = IOBluetoothDeviceInquiryCreateWithCallbackRefCon (&dd);
                    ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    osx/_osxbt.c:705:5: warning: implicit declaration of function 'IOBluetoothDeviceInquirySetCompleteCallback' is invalid in C99 [-Wimplicit-function-declaration]
        IOBluetoothDeviceInquirySetCompleteCallback (dd->inquiry,
        ^
    osx/_osxbt.c:717:5: warning: implicit declaration of function 'IOBluetoothDeviceInquiryStart' is invalid in C99 [-Wimplicit-function-declaration]
        IOBluetoothDeviceInquiryStart (dd->inquiry);
        ^
    osx/_osxbt.c:721:5: warning: implicit declaration of function 'IOBluetoothDeviceInquiryDelete' is invalid in C99 [-Wimplicit-function-declaration]
        IOBluetoothDeviceInquiryDelete (dd->inquiry);
        ^
    5 warnings and 2 errors generated.
    error: command 'gcc' failed with exit status 1
    
    ----------------------------------------
Command "/Users/640labs/anaconda/envs/pybluez/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/bs/0t149_8x579fg8m295g6rvs80000gn/T/pip-build-Dw0d87/pybluez/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/bs/0t149_8x579fg8m295g6rvs80000gn/T/pip-CBAPnW-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/bs/0t149_8x579fg8m295g6rvs80000gn/T/pip-build-Dw0d87/pybluez/
```

vtrubets, 2016-09-14:
I'm seeing the same failure on OSX El Capitan 10.11.6.

jwmillr, 2016-09-20:
I have the same issue on OSX El Capitan 10.11.6. I had tried the following:
gpip install pyobjc // (ran successfully)
gpip install pybluez

creedda, 2016-10-29:
Same issue on OS X Sierra 10.12.1 for pip3.

wamatt, 2016-10-31:
Anyone have a fix yet?

karulis, 2016-10-31:
If You would like to try,there is possible fix in PR
https://github.com/karulis/pybluez/pull/131
But I had no time to test it so it is not yet merged

ToshiyukiAndo, 2016-12-06:
I have same Issue on OS X el Capitan and python3.5.1.
Any one could fix this issue?

Tristan79, 2016-12-23:
It goes away when you install bot pybluez & pyobj 3.2.1 from source. 
But then following error occurs...

```
Traceback (most recent call last):
  File "sdp-browse.py", line 7, in <module>
    import bluetooth
  File "build/bdist.macosx-10.11-x86_64/egg/bluetooth/__init__.py", line 47, in <module>
  File "build/bdist.macosx-10.11-x86_64/egg/bluetooth/osx.py", line 1, in <module>
  File "build/bdist.macosx-10.11-x86_64/egg/lightblue/__init__.py", line 160, in <module>
  File "build/bdist.macosx-10.11-x86_64/egg/lightblue/_lightblue.py", line 31, in <module>
  File "build/bdist.macosx-10.11-x86_64/egg/lightblue/_macutil.py", line 169, in <module>
objc.BadPrototypeError: Objective-C expects 1 arguments, Python argument has 2 arguments for <unbound selector sleep of BBCocoaSleeper at 0x11158bd98>

```

pramsey, 2016-12-25:
I'm seeing the same thing as @Tristan79

emouty, 2017-01-03:
I have the same issue than @Tristan79

garrett-davidson, 2017-01-21:
Currently running macOS 10.12.3 beta, I had the same issue through pip as @mlyons-tcc . However, I was able to download the tarball (`karulis-pybluez-0.22-71-ga0b226a` currently) and install from source. I had an issue where it said that `libffi.h` was not found. Running `brew link libffi --force` fixed this problem, then I was able to `sudo python setup.py install` successfully.

b4be1, 2017-03-07:
To fix the error described by @Tristan79, one has to decorate several methods with @objc.python_method because of the changes to PyObjC in version 3.2 (see [Backward compatibility note](https://pythonhosted.org/pyobjc/core/changelog.html)). A version that compiles is available at this [pull request](https://github.com/karulis/pybluez/pull/146).

johndpope, 2017-09-21:
@b4be1  - it seems that PR above has been successfully merged.  but I'm seeing same problem 

I'm on 
10.13 Beta (17A360a) - high sierra

failed install log
https://gist.github.com/johndpope/40e3574fcc6fa92ef69300da6f2bdb35'

my installation scripts
https://gist.github.com/johndpope/1f8a0f119f94ee1d3f3c98753c744d9a

Actually - looking closer - using the master (not branch)
```
brew install libffi
brew link libffi --force
sudo python setup.py install
```

seems to work fine.

b4be1, 2017-09-21:
@johndpope, I only decorated several methods that I needed in my project and haven't followed on the issue afterwards. If the master branch works fine, then I suppose all methods were fixed there and you can just use it. Otherwise, add the missing decorators as I did in the [pull request](https://github.com/karulis/pybluez/pull/146/files).

rgov, 2017-11-16:
My pull request, #167, improves Mac support. If you want to try it now, just run the following command. You should not need to install anything else.

`pip install --upgrade git+https://github.com/rgov/pybluez.git#egg=pybluez`

bretcj7, 2017-12-19:
@rgov The command you gave works great for me, no issues installing.  Looks like when running the ble scan test i get the common gattlib error though.

`Traceback (most recent call last):
  File "scan.py", line 1, in <module>
    from bluetooth.ble import DiscoveryService
  File "/usr/local/lib/python3.6/site-packages/bluetooth/ble.py", line 1, in <module>
    from gattlib import *
ModuleNotFoundError: No module named 'gattlib'`

rgov, 2017-12-20:
Hi @bretcj7, please open another bug with the output of `pip freeze`. I'm going to mark this one as fixed.

closed 2017-12-20 by rgov

dongdong3344, 2017-12-29:
@rgov  I used the command `pip install --upgrade git+https://github.com/rgov/pybluez.git#egg=pybluez` you gave to install `PyBluez` module on mac OS with a success.Then I restarted my mac and then import bluetooth  in the IDE PyCharm, unfortunately, it throws error info `ModuleNotFoundError: No module named 'bluetooth'`.It's so weird that because the `PyBluez`module has been in the  package list.Any idea about this ?Thanks.

binlaniua, 2018-03-01:
@rgov it's work, thanks

unixb0y, 2019-05-12:
Unfortunately not working :/

```
$ pipenv run python socketServer.py 
Traceback (most recent call last):
  File "socketServer.py", line 14, in <module>
    import bluetooth
  File "/Users/me/.local/share/virtualenvs/BTtests-ip9bA9ps/lib/python3.7/site-packages/bluetooth/__init__.py", line 47, in <module>
    from .osx import *
  File "/Users/me/.local/share/virtualenvs/BTtests-ip9bA9ps/lib/python3.7/site-packages/bluetooth/osx.py", line 1, in <module>
    import lightblue
  File "/Users/me/.local/share/virtualenvs/BTtests-ip9bA9ps/lib/python3.7/site-packages/lightblue/__init__.py", line 160, in <module>
    from ._lightblue import *
  File "/Users/me/.local/share/virtualenvs/BTtests-ip9bA9ps/lib/python3.7/site-packages/lightblue/_lightblue.py", line 29, in <module>
    from . import _LightAquaBlue
  File "/Users/me/.local/share/virtualenvs/BTtests-ip9bA9ps/lib/python3.7/site-packages/lightblue/_LightAquaBlue.py", line 33, in <module>
    _FRAMEWORK_PATH)
ImportError: Cannot load LightAquaBlue framework, not found at /Users/me/.local/share/virtualenvs/BTtests-ip9bA9ps/lib/python3.7/site-packages/lightblue/LightAquaBlue.framework
```

Python 3.7.3, macOS 10.14.3, installed PyObjC and PyBluez like so:

`pipenv install pyobjc`
`piping install git+https://github.com/rgov/pybluez.git#egg=pybluez`