pybluez/pybluez #135: Missing features and D-Bus bindings

FynnMaz, 2016-07-20:
Hey everybody,

at the moment pybluez is essentially the **only** compatible and up-to-date bluetooth library. 
It does a great job at everything it has implemented and even provides this functionality to Python 2.x and 3.x. 

I am aware that this project probably never intended to be an official and fully developed BlueZ-Wrapper,
however, it has just too little functionality considering it is the go-to python bluetooth library.
There are missing features such as:

1. Being able to read/write adapter settings (D-Bus org.bluez.AdapterX)
2. Implementing a way to set up an A2DP Sink or an RCT.
3. Further low-level bindings exposed at _org.bluez_.

I would like to self-assign this issue since the company I am working for needs a Python Bluetooth library for one of their products. I will try to give pyBlueZ an overhaul. 
It would be great if you could just give me some heads-up on tricky parts or hacky paths that you took, so it is easier for me to wrap my head around the whole thing.

Cheers.

karulis, 2016-07-20:
Hi Fynn,

Every help to keep pybluez up to date is very appreciated.
I believe there are no major hacks - but if You found any problem during Your development i will more then happy to help:)

And by the way - keep in mind that pybluez tries to be corss platform library.
But off course it is better to have some feature supported only on linux rather dont have them at all.

FynnMaz, 2016-07-20:
Okay, I would love to help and give something back to the community. Obviously, since its your project, I don't want to use my own coding style and rather stick close to what you already established as standard for this one. Is there anything you want me to remember in this respect?

For example: 
Where would I implement functions that mimic the functionality of `bluetoothctl` (packed with BlueZ 5.x)? Would you rather want them all to be functions in the _bluetooth_ package (as a high-level memeber) or have them in a new `Adapter` class in the form of `bluetooth.getAdapter().setDiscoverable()`? 

Basically, OOP or functional? I (sadly) come from an OOP background and therefore tend to write classes for everything, which limits my perspective but increases code readability (I think). I already have some D-Bus bindings done and will start implementing them in a fork soon.

FynnMaz, 2016-07-28:
My work on this project is scheduled for the 8th of August. Could you just take some time to answer my questions until then? :)

karulis, 2016-07-28:
Hi Fynn,

Sorry for not very regular responses:)

> Where would I implement functions that mimic the functionality of bluetoothctl

https://github.com/karulis/pybluez/tree/master/bluetooth contains(mostly) high level pybluez API functions
If the implementation is complex I would move it to https://github.com/karulis/pybluez/tree/master/bluez (as is in case of osx implementation)

> Would you rather want them all to be functions in the bluetooth package (as a high-level memeber) or have them in a new Adapter class in the form of bluetooth.getAdapter().setDiscoverable()? 

Your implementation will be bluez 5.x based - so we need some nice and gantle way to inform users of older versions that they are not supported or maybe just not import additional implementation.
Anyway It would be best to keep bluez 5.x implementation in separate module/source file

> Basically, OOP or functional?

I  believe OOP is preferedd way

Main things to think during development:
- backward compatybility - so phisical separation of bluez 5.x implementation
- make it as cross platform as posible(so that we can add support for windows/osx with same API) - try to hide bluez-only/linux things, and maybe consult from time to time other platforms documentation

And BTW, this is not my project:)
Albert Huang is the creator, I happen to be maintaining it - but this is open source community's project:)

karulis, 2016-08-25:
@FynnMaz how goes Your implementation?

FynnMaz, 2016-08-30:
Hey Karulis,

at the moment my focus shifted and blue-tooth is no longer important for prototyping.
I will get back to this if priorities change again or find some spare time, both being rather unlikely.
Nonetheless, we managed to solve our problem with the help of your library so thank you for supplying
such a great nifty module.

Cheers

karulis, 2016-10-31:
Thanks, closing then:)

closed 2016-10-31 by karulis

dude8604, 2017-07-23:
Does anyone else want to add this functionality? It would be very helpful for my project, but I don't really have the knowledge to implement it myself. Thanks.

FynnMazurkiewicz, 2017-07-25:
What kind of functionality do you aim to achieve? Actually, I managed to include PyBluez into my last project and filled in some missing features with my own.

dude8604, 2017-07-26:
I'm stuck on two things. One is how to set up an agent to programmatically decide whether to accept an incoming pairing request. I found some code to do it, but it uses the dbus interface directly. It looks like it would be complicated to learn about dbus and how to use it. https://github.com/pauloborges/bluez/blob/master/test/simple-agent

The other thing, and actually there might be a way to do this with PyBluez, is to have the computer act as a peripheral device. I want to have the computer act as a mouse and keyboard for the host device.

If you know how to do either of these things, please let me know. I would also consider paying someone to program this if necessary, either for me, or ideally to add the functionality to PyBluez. Thanks.

FynnMazurkiewicz, 2017-07-28:
They correct way to setting up an agent depends on the version of BlueZ that you want to use. The new versions shifted their entire control to the DBus-API. Which version do you plan on using?

As for the second step, it likely won't be possible unless you plan to roll out your own/a custom android.
Vanilla android implements these [profiles](https://developer.android.com/guide/topics/connectivity/bluetooth.html#Profiles). As you can see, the HID-profile isn't one of them, which means that, even if you could tell another device to acts a peripheral, Android wouldn't understand.

One solution would be to roll out a custom serial protocol and interpret received commands with virtual inputs on the host. I am not a sufficiently good Android developer though, so I can't tell how or if this can be done.

FynnMazurkiewicz, 2017-08-03:
Actually, I would like to correct myself. It seems that [this works](https://www.youtube.com/watch?v=T1PCpXbf_kU) on (some?) distributions. Sorry!