pybluez/pybluez #140: maybe https://pypi.python.org/pypi/PyBluez too old?

shimondoodkin, 2016-11-22:
https://pypi.python.org/pypi/PyBluez has version from 2015
that does not contain local device address

I wish that nfc-bluetooth examples would work out of the box on Windows and Linux
https://github.com/nfcpy/nfcpy/blob/master/examples/handover-test-client.py
https://github.com/nfcpy/nfcpy/blob/master/examples/handover-test-server.py

karulis, 2016-11-23:
I dont have much time right now for new release,
I plan to release it after python 3.6 release (3.6.0 final: 2016-12-16 )
so maybe by Christmas.

RyuzakiKK, 2017-07-11:
@karulis do you have any news about that?

vit1251, 2017-12-08:
I see on PyPi server actual version is 0.22 but does not understand current version and News page.

closed 2019-07-30 by stale[bot]