pybluez/pybluez #142: Error install Ubuntu 16.04

alejosv, 2016-12-13:
When I try install in Ubuntu 16.04 64 bits have this error:

Command: `sudo python3 setup install`

`creating build/temp.linux-x86_64-3.5/bluez
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I./port3 -I/usr/include/python3.5m -c bluez/btmodule.c -o build/temp.linux-x86_64-3.5/bluez/btmodule.o
In file included from bluez/btmodule.c:20:0:
bluez/btmodule.h:5:33: fatal error: bluetooth/bluetooth.h: No existe el archivo o el directorio
compilation terminated.
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1`

And if I use this command:
`pip3 install pybluez`

Have the next issue:

`Collecting pybluez
  Using cached PyBluez-0.22.zip
Building wheels for collected packages: pybluez
  Running setup.py bdist_wheel for pybluez ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-jg3yssmw/pybluez/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp_7y30pv7pip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.5
  creating build/lib.linux-x86_64-3.5/bluetooth
  copying bluetooth/widcomm.py -> build/lib.linux-x86_64-3.5/bluetooth
  copying bluetooth/msbt.py -> build/lib.linux-x86_64-3.5/bluetooth
  copying bluetooth/__init__.py -> build/lib.linux-x86_64-3.5/bluetooth
  copying bluetooth/ble.py -> build/lib.linux-x86_64-3.5/bluetooth
  copying bluetooth/bluez.py -> build/lib.linux-x86_64-3.5/bluetooth
  copying bluetooth/osx.py -> build/lib.linux-x86_64-3.5/bluetooth
  copying bluetooth/btcommon.py -> build/lib.linux-x86_64-3.5/bluetooth
  running build_ext
  building 'bluetooth._bluetooth' extension
  creating build/temp.linux-x86_64-3.5
  creating build/temp.linux-x86_64-3.5/bluez
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I./port3 -I/usr/include/python3.5m -c bluez/btmodule.c -o build/temp.linux-x86_64-3.5/bluez/btmodule.o
  In file included from bluez/btmodule.c:20:0:
  bluez/btmodule.h:5:33: fatal error: bluetooth/bluetooth.h: No existe el archivo o el directorio
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for pybluez
  Running setup.py clean for pybluez
Failed to build pybluez
Installing collected packages: pybluez
  Running setup.py install for pybluez ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-jg3yssmw/pybluez/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-w0qswhzg-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.5
    creating build/lib.linux-x86_64-3.5/bluetooth
    copying bluetooth/widcomm.py -> build/lib.linux-x86_64-3.5/bluetooth
    copying bluetooth/msbt.py -> build/lib.linux-x86_64-3.5/bluetooth
    copying bluetooth/__init__.py -> build/lib.linux-x86_64-3.5/bluetooth
    copying bluetooth/ble.py -> build/lib.linux-x86_64-3.5/bluetooth
    copying bluetooth/bluez.py -> build/lib.linux-x86_64-3.5/bluetooth
    copying bluetooth/osx.py -> build/lib.linux-x86_64-3.5/bluetooth
    copying bluetooth/btcommon.py -> build/lib.linux-x86_64-3.5/bluetooth
    running build_ext
    building 'bluetooth._bluetooth' extension
    creating build/temp.linux-x86_64-3.5
    creating build/temp.linux-x86_64-3.5/bluez
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I./port3 -I/usr/include/python3.5m -c bluez/btmodule.c -o build/temp.linux-x86_64-3.5/bluez/btmodule.o
    In file included from bluez/btmodule.c:20:0:
    bluez/btmodule.h:5:33: fatal error: bluetooth/bluetooth.h: No existe el archivo o el directorio
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-jg3yssmw/pybluez/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-w0qswhzg-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-jg3yssmw/pybluez/`

karulis, 2016-12-13:
Have You tried solution to issue #75 ?

karulis, 2016-12-25:
duplicate of issue #75

closed 2016-12-25 by karulis