pybluez/pybluez #144: Issue importing bluetooth on OSX

mhassell, 2017-02-07:
Good evening,  I just installed pybluez and attempt to run import bluetooth in a python shell, I get the following errors:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "build/bdist.macosx-10.11-intel/egg/bluetooth/__init__.py", line 47, in <module>
  File "build/bdist.macosx-10.11-intel/egg/bluetooth/osx.py", line 1, in <module>
  File "build/bdist.macosx-10.11-intel/egg/lightblue/__init__.py", line 160, in <module>
  File "build/bdist.macosx-10.11-intel/egg/lightblue/_lightblue.py", line 31, in <module>
  File "build/bdist.macosx-10.11-intel/egg/lightblue/_macutil.py", line 169, in <module>
objc.BadPrototypeError: Objective-C expects 1 arguments, Python argument has 2 arguments for <unbound selector sleep of BBCocoaSleeper at 0x111c68100>

If I persist and try importing again, I get the following error:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "build/bdist.macosx-10.11-intel/egg/bluetooth/__init__.py", line 47, in <module>
  File "build/bdist.macosx-10.11-intel/egg/bluetooth/osx.py", line 1, in <module>
  File "build/bdist.macosx-10.11-intel/egg/lightblue/__init__.py", line 160, in <module>
  File "build/bdist.macosx-10.11-intel/egg/lightblue/_lightblue.py", line 28, in <module>
ImportError: cannot import name _IOBluetooth

Any advice would be appreciated.

mhassell, 2017-02-09:
Sorry, realized that this isn't compatible with OSX.

closed 2017-02-09 by mhassell

rgov, 2017-12-21:
@mhassell It should be more compatible with macOS now, if you want to try.