pybluez/pybluez #148: Raspbian install fail

fiveseven808, 2017-03-29:
```
pi@hueBerry ~/pybluez> sudo python setup.py install
running install
running bdist_egg
running egg_info
writing requirements to PyBluez.egg-info/requires.txt
writing PyBluez.egg-info/PKG-INFO
writing top-level names to PyBluez.egg-info/top_level.txt
writing dependency_links to PyBluez.egg-info/dependency_links.txt
reading manifest file 'PyBluez.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching '*.c' under directory 'osx'
warning: no previously-included files matching '*.pyc' found under directory '*'
warning: no previously-included files matching '*.o' found under directory '*'
warning: no previously-included files matching '*.opp' found under directory '*'
writing manifest file 'PyBluez.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv6l/egg
running install_lib
running build_py
running build_ext
building 'bluetooth._bluetooth' extension
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I./port3 -I/usr/include/python2.7 -c bluez/btmodule.c -o build/temp.linux-armv6l-2.7/bluez/btmodule.o
In file included from bluez/btmodule.c:20:0:
bluez/btmodule.h:5:33: fatal error: bluetooth/bluetooth.h: No such file or directory
 #include <bluetooth/bluetooth.h>
                                 ^
compilation terminated.
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```
I looked around at other issues and it seems I have a good enough version of bluez, no? 

```
pi@hueBerry ~/pybluez> apt-cache policy bluez
bluez:
  Installed: 5.23-2+rpi2
  Candidate: 5.23-2+rpi2
  Version table:
 *** 5.23-2+rpi2 0
        500 http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status
     5.23-2+b1 0
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
```

Any help would be greatly appreciated. Thanks!

GozzoMan, 2017-03-31:
It seems to install after installing:
sudo apt-get install libbluetooth-dev

(Take it with a grain of salt: I still haven't my code running tough.)

camilleeyries, 2017-04-24:
@GozzoMan thanks, working.

bretcj7, 2017-08-25:
GozzoMan steps worked for me too on pi 3 running jessie.  Thanks!

closed 2019-07-30 by stale[bot]