pybluez/pybluez #149: Thread Blocking while scanning for BLE devices

tobiasfeistmantl, 2017-04-02:
Hi.

If I use the [BLE scan script](https://github.com/karulis/pybluez/blob/master/examples/ble/scan.py) in an infinite loop and put it into another thread with threading, it blocks the other threads while scanning. After every round some of the other operations of the other threads are executed.

Here is my code:

```python
from bluetooth.ble import DiscoveryService
import threading

def run():
  service = DiscoveryService()
  while True:
    devices = service.discover(2)
    for address, name in devices.items():
      print("name: {}, address: {}".format(name, address))

discovery_loop = threading.Thread(target=run)
discovery_loop.start()
```

Thank you in advance.

IndyJoeA, 2017-07-23:
I experienced this same problem.  My workaround was to switch from multiple threads to multiple processes.  https://docs.python.org/2/library/multiprocessing.html

kaskavalci, 2017-08-20:
@IndyJoeA Can you post an example?

vit1251, 2017-08-24:
It look like C binding code acquire GIL lock and does not use Py_BEGIN_ALLOW_THREADS macros.
You can see more on https://docs.python.org/2/c-api/init.html

closed 2019-07-30 by stale[bot]