pybluez/pybluez #151: Raspbian pip install fails - Python.h: no such file or directory

JBaczuk, 2017-05-13:
Installation fails using `pip install pybluez` with the following error:

```
bluez/btmodule.h:4:20: fatal error: Python.h: No such file or directory

 #include "Python.h"

                    ^

compilation terminated.

error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

----------------------------------------
  Can't roll back PyBluez; was not uninstalled
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-tEa28i/pybluez/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-usS71K-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-tEa28i/pybluez
Storing debug log for failure in /home/pi/.pip/pip.log
```

otoomey, 2017-05-15:
I am running Raspbian version 4.4.50+, and I was able to install it successfully. Make sure you have the relevant headers required for `pybluez` installed.

Also try running it with `sudo`. it might be a privileges issue. Nonetheless, the problem is with your configuration, not with the `pybluez` it self.

KinmanLam, 2017-05-19:
@JBaczuk  you are missing the Python Dev libraries. You will also need to install the Bluetooth Dev libraries.

I wrote a guide to get PyBlueZ running on a Raspberry Pi. Check it out: http://www.kinmanlam.com/2017/05/the-5-minute-guide-to-setup-raspberry.html

hashans, 2017-07-21:
@KinmanLam provided link really helped. Was able to successfully install pybluez on my Raspberry Pi. Thanks.

danielunderwood, 2018-03-13:
The answer above is correct. You just need a `sudo apt-get install python-dev`.

rgov, 2018-03-14:
A pull request to document this prerequisite in the README would be appreciated. I'll close this issue since it seems to be resolved.

closed 2018-03-14 by rgov