pybluez/pybluez #157: Install pybluez under virtualenv

sisihi, 2017-07-05:
Hi,
I've successfully install the pybluez with the global python environment, but failed after activate the virtualenv:

Installing collected packages: pybluez
  Running setup.py install for pybluez ... error
    Complete output from command /home/szp/virtualenv_2017_06_23/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-dIFQJi/pybluez/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-LdFcqm-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/szp/virtualenv_2017_06_23/include/site/python2.7/pybluez:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-2.7
    creating build/lib.linux-x86_64-2.7/bluetooth
    copying bluetooth/widcomm.py -> build/lib.linux-x86_64-2.7/bluetooth
    copying bluetooth/ble.py -> build/lib.linux-x86_64-2.7/bluetooth
    copying bluetooth/btcommon.py -> build/lib.linux-x86_64-2.7/bluetooth
    copying bluetooth/osx.py -> build/lib.linux-x86_64-2.7/bluetooth
    copying bluetooth/msbt.py -> build/lib.linux-x86_64-2.7/bluetooth
    copying bluetooth/__init__.py -> build/lib.linux-x86_64-2.7/bluetooth
    copying bluetooth/bluez.py -> build/lib.linux-x86_64-2.7/bluetooth
    running build_ext
    building 'bluetooth._bluetooth' extension
    creating build/temp.linux-x86_64-2.7
    creating build/temp.linux-x86_64-2.7/bluez
    x86_64-linux-gnu-gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I./port3 -I/usr/include/python2.7 -c bluez/btmodule.c -o build/temp.linux-x86_64-2.7/bluez/btmodule.o
    In file included from bluez/btmodule.c:20:0:
    bluez/btmodule.h:5:33: fatal error: bluetooth/bluetooth.h: No such file or directory
     #include <bluetooth/bluetooth.h>
                                     ^
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

Did I miss something?

Thanks for the help.

yangoff, 2017-07-18:
sudo apt-get install libbluetooth-dev bluez bluez-hcidump  libboost-python-dev libboost-thread-dev libglib2.0-dev

jslay88, 2017-11-30:
You only need python-dev (for python 2) or python3-dev (for python 3), and libbluetooth-dev.

rgov, 2017-12-20:
Is this still an issue for you @sisihi?

banjin, 2018-01-24:
apt-get install libbluetooth-dev

JimmyTheNerd, 2018-11-16:
`sudo apt-get install libbluetooth-dev` worked for me.

closed 2019-07-30 by stale[bot]

mazunki, 2019-08-03:
On SolusOS `libbluetooth` is called `bluez`, so the solution would be `sudo eopkg it bluez-devel`. You also need to install `python3-devel`, to solve the `Python.h` missing file.