pybluez/pybluez #158: Can device discovery be cancelled?

pkozul, 2017-07-10:
From what I can tell, the DeviceDiscoverer is not implemented in the Windows version. I was hoping to use the DeviceDiscoverer because it supports cancelling the inquiry.

Using the discover_devices() function to kick of discovery, is there a way to cancel the discovery?

muelli, 2019-07-29:
This is also interesting for us.

muelli, 2019-08-28:
this is pathetic.

rgov, 2019-08-29:
@muelli I don't know how we can be more clear about it. No one is developing PyBluez. It says it in the README, it says it when you file an issue, the bot says it when no one responds.

It looks like from your GitHub history that you are a developer, so why don't you take a stab at implementing this? The maintainers can do a code review and merge it in for you.

If you would rather not spend time developing features for PyBluez, that's fine—neither does anyone else, which is why no one does. But please don't post "me too" comments to issues, especially derisive ones that don't help anyone.

closed 2019-10-05 by stale[bot]