pybluez/pybluez #160: macOS issue when using pybluez in separate thread

bmesuere, 2017-07-30:
When using pybluez in a separate thread (which for example Home Assistant does) on macOS, the code throws an error:

```
...
  File "/usr/local/Cellar/python3/3.6.2/Frameworks/Python.framework/Versions/3.6/lib/python3.6/concurrent/futures/thread.py", line 55, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/lib/python3.6/site-packages/homeassistant/components/device_tracker/bluetooth_tracker.py", line 60, in setup_scanner
    for dev in discover_devices():
  File "/lib/python3.6/site-packages/homeassistant/components/device_tracker/bluetooth_tracker.py", line 38, in discover_devices
    lookup_class=False)
  File "/usr/local/lib/python3.6/site-packages/PyBluez-0.22-py3.6.egg/bluetooth/osx.py", line 12, in discover_devices
    devices = lightblue.finddevices(getnames=lookup_names, length=duration)
  File "/usr/local/lib/python3.6/site-packages/PyBluez-0.22-py3.6.egg/lightblue/_lightblue.py", line 48, in finddevices
    inquiry.run(getnames, length)
  File "/usr/local/lib/python3.6/site-packages/PyBluez-0.22-py3.6.egg/lightblue/_lightblue.py", line 349, in run
    _macutil.waituntil(lambda: not self._inquiring)
  File "/usr/local/lib/python3.6/site-packages/PyBluez-0.22-py3.6.egg/lightblue/_macutil.py", line 152, in waituntil
    e = app.nextEventMatchingMask_untilDate_inMode_dequeue_(NSAnyEventMask, NSDate.dateWithTimeIntervalSinceNow_(min(endtime - currtime, WAIT_MAX_TIMEOUT)), NSDefaultRunLoopMode, True)
objc.error: NSInternalInconsistencyException - nextEventMatchingMask should only be called from the Main Thread!
```

I am using the latest code from github.

pxxgogo, 2019-05-13:
@bmesuere Hi! I met the same problem. Do you have any solutions?

bmesuere, 2019-05-16:
No, sorry, I have given up on using Bluetooth on my mac :)

pxxgogo, 2019-05-19:
I realized it with multi-process, which is not bad up to now.

closed 2019-07-30 by stale[bot]