pybluez/pybluez #161: rfcomm-server.py just fails with "No such file or directory"

skoehler, 2017-08-08:
The rfcomm-server example just fails. The error message is confusing at best. I don't know what's going on, I even tried running the script as root.

I'm using python 3.4 and pybluez 0.22.

Bluetooth is working just fine with other applications. rfcomm kernel module is loaded.

```
$ python3 rfcomm-server.py 
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/bluetooth/bluez.py", line 240, in advertise_service
    protocols)
_bluetooth.error: (2, 'No such file or directory')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "rfcomm-server.py", line 20, in <module>
    profiles = [ SERIAL_PORT_PROFILE ], 
  File "/usr/lib64/python3.4/site-packages/bluetooth/bluez.py", line 242, in advertise_service
    raise BluetoothError (str (e))
bluetooth.btcommon.BluetoothError: (2, 'No such file or directory')
```

infernoDison, 2017-08-11:
The serial port profile is missing? Check the post [Bluetooth Missing SPP](https://www.raspberrypi.org/forums/viewtopic.php?t=133263&p=887944). This post is for linux system on Raspberry Pi, but could be helpful to the solution.

Damncutecat, 2017-08-28:
I have encountered such a problem, I have modified 
sudo vi /etc/systemd/system/dbus-org.bluez.service
The But can not solve the problem.
我也遇到这样的问题，我已经修改sudo vi /etc/systemd/system/dbus-org.bluez.service
。但不能解决问题。

zndemao, 2017-08-28:
I found a solution. Do not know can solve the problem, I have not down to test. There will not be such a mistake.
The workaround is to add the following to / etc / bluetooth / main.conf:
    DisablePlugins = pnat
Reference: http://blog.davidvassallo.me/2014/05/11 ... unication/

我找到了一个解决方案。不知道能不能解决问题，我还没有往下进行测试。现在不会出现这样的错误。
解决方法是将以下内容添加到/ etc / bluetooth / main.conf中：
DisablePlugins = pnat
参考:http://blog.davidvassallo.me/2014/05/11 ... unication/

I'm not sure if this method is available
我不确定这个方法是否可以

heesukahn, 2017-10-02:
I recommend this solution

https://raspberrypi.stackexchange.com/questions/41776/failed-to-connect-to-sdp-server-on-ffffff000000-no-such-file-or-directory

Maroju100, 2017-10-11:
You can solve by enabling compatibility mode and loading the serial profile by editing file /lib/systemd/system/bluetooth.service and changing line:

ExecStart=/usr/lib/bluetooth/bluetoothd
to:

ExecStart=/usr/lib/bluetooth/bluetoothd -C
and then running sudo sdptool add SP.

http://howcanfix.com/81047/communicating-via-bluetooth-serial-with-python

meeuw, 2018-02-03:
Just adding `-C` or `--compat` to `bluetoothd` seems to be work for me. Does someone know how this should be fixed without using compatibilty mode?

Cheaterman, 2019-03-09:
This is still an issue... adding `-C` does work, but it's not a solution when you realize this SDP socket interface has been deprecated for SEVEN YEARS now... Please fix!

rgov, 2019-03-09:
@cheaterman As the project readme says, there are no active developers. If you can contribute a change, I can merge it in.

closed 2019-07-30 by stale[bot]

Phlip, 2022-01-31:
just a complaint; the error comes back each time an `apt upgrade` refreshes the `bluetooth.service` file. )-: