pybluez/pybluez #162: Device or resource busy

SeanMaday, 2017-08-21:
I am trying to poll a Garmin Glo once per second from a Raspberry Pi 3 over Bluetooth and I am frequently getting a "Device or resource busy" error. My code is below. Is there a better way to architect this workflow?

```
#!/usr/bin/env python
# encoding: utf-8

import time
import json
import pynmea2
import bluetooth

def getLocation():
	bdAddr = "10:C6:FC:6B:13:A3"
	bdPort = 1
	sockObj = bluetooth.BluetoothSocket(bluetooth.RFCOMM)
	sockObj.connect((bdAddr, bdPort))
	rawData = sockObj.recv(1024)
	sockObj.close()

	nmeaMsg = pynmea2.parse(rawData)

	outpuObj = {}
	outpuObj["rxLat"] = nmeaMsg.latitude
	outpuObj["rxLon"] = nmeaMsg.longitude
	outpuObj["rxStatus"] = nmeaMsg.status
	outpuObj["rxDatestamp"] = str(nmeaMsg.datestamp)
	outpuObj["rxTimestamp"] = str(nmeaMsg.timestamp)

	dataDir = "/home/pi"
	filePath = dataDir + "/" + "currentLocation.json"
	open(filePath, "wb").write(json.dumps(outpuObj))

	return outpuObj

def main():
        while True:
                try:
                        print getLocation()
                        time.sleep(1)

                except KeyboardInterrupt:
                        quit()

                except Exception,e:
                        print e
                        pass

if __name__ == "__main__":
	main()
```

FynnMazurkiewicz, 2017-09-01:
The reason that you might be getting these is are the quick successive calls to `BluetoothSocket()` and `BluetoothSocket().close()`. If the clean-up of the `close`-call takes longer than your 1 second sleep and the `write()`, your BluetoothDevice will still be blocked when you try to reopen the socket. Consider creating it and connecting once, leaving it open and only using `recv` in `getLocation()`.

rgov, 2017-12-20:
Did that fix the issue for you, @SeanMaday ?

closed 2019-07-30 by stale[bot]