pybluez/pybluez #163: Installation of BLE

360disrupt, 2017-09-15:
I installed first:
```
sudo apt-get install pi-bluetooth
sudo apt-get install bluetooth bluez
sudo apt-get install bluez python-bluez
pip install pybluez
```

then I installed the [dependencies](https://bitbucket.org/OscarAcena/pygattlib/src/45e04060881a20189412681f52d55ff5add9f388/DEPENDS?at=default) for BLE
`sudo apt-get install libboost-python-dev libboost-thread-dev libbluetooth-dev libglib2.0-dev python-dev` pkg-config was already available.

At last I tried to install the additional dep.
`pip install pybluez[ble]`

`PyBluez 0.18 does not provide the extra 'ble'`

Running the example:
```
  File "r.py", line 23, in <module>
    from bluetooth.ble import DiscoveryService
ImportError: No module named ble
```

wb666greene, 2017-09-22:
It worked for me with Raspbian-stretch 2017-09-07 doing:
sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install libboost-python-dev libboost-thread-dev libbluetooth-dev libglib2.0-dev
sudo pip install pybluez
sudo pip install pybluez[ble]
sudo nano /lib/systemd/system/bluetooth.service
// then add --experimental to the ExecStart= line so it looks something like this:
ExecStart=/usr/lib/bluetooth/bluetoothd --experimental

Then reboot or restart the Bluetooth service.

Unfortunately none of the ble examples did anything when I ran them.  No errors were thrown, but the scan didn't find my two Radius Dot beacons or any other Bluetooth devices, so I moved on to an alternate kludge using hcitool lescan from a script.

360disrupt, 2017-09-23:
I bought a dongle and my BLE worked. I assume that the pi Bluetooth is just not working (on my board).

YMoserAtLogi, 2019-01-08:
I had the same issue when running 
`sudo pip install bluez[ble]`
`PyBluez 0.18 does not provide the extra 'ble'`

 Tried a lot of things and seems that dropping the "sudo" makes it work. Simply run this:
`pip install bluez[ble]`

 Hope this helps.

closed 2019-07-30 by stale[bot]