pybluez/pybluez #165: macOS 10.13 Exception NSInvalidArgumentException

andrrra, 2017-09-29:
All that I am doing is 

```
import bluetooth
nearby_devices = bluetooth.discover_devices(lookup_names=True)
```

and whenever there's a bluetooth device nearby that can be connected, I get
```
2017-09-29 16:26:22.327 Python[12172:429449] Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
2017-09-29 16:26:22.327 Python[12172:429449] Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
2017-09-29 16:26:22.328 Python[12172:429449] Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
2017-09-29 16:26:22.329 Python[12172:429449] Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
```

If I turn the device off and on again the four lines print out again. The messages don't seem to be preventing me from establishing a connection though. Is this the intended behaviour? I have attempted making the connection with **lightblue** as well, but I had the same problem (but then I looked through the osx source code for **pybluez** and noticed that it is actually based on lightblue). 

I should note that I did not have this issue before upgrading to macOS 10.13. 
And thank you for taking the time to read this :)

fipstree, 2017-10-01:
Got exactly the same error.
Always saying 0 devices found. :-(
Already tried to fix it myself, but can't figure out what is wrong. It seems like Apple made some changes to Bluetooth API since latest OS Update?

aj-ptw, 2017-10-28:
Yes they did see [`noble/#689`](https://github.com/sandeepmistry/noble/pull/689)

rgov, 2017-12-20:
macOS support should be improved now, please try installing from `master` and tell us if it still is an issue.

dongdong3344, 2018-01-04:
@rgov 'installing from master' ? what does it mean  ? (I got the same error)

closed 2019-07-30 by stale[bot]