pybluez/pybluez #170: unable to install pybluez under raspberry pi

ag3ntsim0, 2017-11-29:
unable to install this module under my raspberry pi 3 !

after running pip i get this message error
`compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-2.7
    creating build/lib.linux-armv7l-2.7/bluetooth
    copying bluetooth/msbt.py -> build/lib.linux-armv7l-2.7/bluetooth
    copying bluetooth/osx.py -> build/lib.linux-armv7l-2.7/bluetooth
    copying bluetooth/__init__.py -> build/lib.linux-armv7l-2.7/bluetooth
    copying bluetooth/btcommon.py -> build/lib.linux-armv7l-2.7/bluetooth
    copying bluetooth/bluez.py -> build/lib.linux-armv7l-2.7/bluetooth
    copying bluetooth/ble.py -> build/lib.linux-armv7l-2.7/bluetooth
    copying bluetooth/widcomm.py -> build/lib.linux-armv7l-2.7/bluetooth
    running build_ext
    building 'bluetooth._bluetooth' extension
    creating build/temp.linux-armv7l-2.7
    creating build/temp.linux-armv7l-2.7/bluez
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I./port3 -I/usr/include/python2.7 -c bluez/btmodule.c -o build/temp.linux-armv7l-2.7/bluez/btmodule.o
    In file included from bluez/btmodule.c:20:0:
    bluez/btmodule.h:5:33: fatal error: bluetooth/bluetooth.h: No such file or directory
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-2aB167/pybluez/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-95Ugz7-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-2aB167/pybluez/`

ag3ntsim0, 2017-11-29:
and this one for pip install pybluez[ble]

compile:
    running install
    running build
    running build_ext
    building 'gattlib' extension
    creating build
    creating build/temp.linux-armv7l-2.7
    creating build/temp.linux-armv7l-2.7/src
    creating build/temp.linux-armv7l-2.7/src/bluez
    creating build/temp.linux-armv7l-2.7/src/bluez/lib
    creating build/temp.linux-armv7l-2.7/src/bluez/attrib
    creating build/temp.linux-armv7l-2.7/src/bluez/src
    creating build/temp.linux-armv7l-2.7/src/bluez/src/shared
    creating build/temp.linux-armv7l-2.7/src/bluez/btio
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/gattservices.cpp -o build/temp.linux-armv7l-2.7/src/gattservices.o
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
    In file included from src/gattservices.cpp:12:0:
    src/gattlib.h:11:33: fatal error: boost/python/list.hpp: No such file or directory
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-dyUNGo/gattlib/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-jRBwRY-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-dyUNGo/gattlib/

jslay88, 2017-11-30:
You need to install the library packages.

sudo apt install python-dev (for python 2)
sudo apt install python3-d (for python 3)
sudo apt install libbluetooth-dev

ag3ntsim0, 2017-11-30:
thx , i have installed the original pybluez after installing libbluetooth-dev!
but now i still cannot install pybluez[ble] !!

also i cannot find python3-dev !

sudo apt-get install python3-d
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package python3-d

ag3ntsim0, 2017-12-02:
well after installing these dependencies now i have installed pybluez[ble] successfully
https://bitbucket.org/OscarAcena/pygattlib/src/45e04060881a20189412681f52d55ff5add9f388/DEPENDS?at=default&fileviewer=file-view-default

rgov, 2017-12-20:
Based on your last message, closing this.

closed 2017-12-20 by rgov

sarosh, 2018-02-02:
https://stackoverflow.com/questions/41463847/got-error-while-download-gattlib-via-pip3 worked for me on Stretch, Python 3.5

timdonovanuk, 2019-06-17:
Thanks...still an issue 18 months on!

rgov, 2019-06-17:
@timdonovanuk There is no one currently working on PyBluez. If you are capable of digesting the instructions from the links in this thread and creating a pull request that would address the issue, I can merge it in. Thanks.