pybluez/pybluez #173: Installation dependency broken on Python 3

vit1251, 2017-12-08:
When use 

```pip install pybluez```

And start using example about low energy example, we have:

```
Traceback (most recent call last):
  File "./b2.py3", line 7, in <module>
    from bluetooth.ble import DiscoveryService
  File "/usr/local/lib/python3.5/dist-packages/bluetooth/ble.py", line 1, in <module>
    from gattlib import *
ImportError: No module named 'gattlib'
```

Require additional install dependencies.

FynnMazurkiewicz, 2017-12-23:
Confirmed on Ubuntu 16.04. This is definitely down to gattlib.
Following [their own install instructions](https://bitbucket.org/OscarAcena/pygattlib/src/45e04060881a?at=default) doesn't work either.
 The repository installation also failed, I haven't tried a source install.

Catshy, 2019-03-26:
install these depends first,
```
pkg-config
libboost-python-dev
libboost-thread-dev
libbluetooth-dev >= 4.101
libglib2.0-dev
python-dev
```
then, for gattlib with python3, have a try with :
```
pip3 download gattlib
tar xvzf ./gattlib-0.20150805.tar.gz
cd gattlib-0.20150805/
sed -ie 's/boost_python-py34/boost_python-py35/' setup.py
pip3 install .
``` 
good luck! 👍

asaladino, 2019-07-14:
I using ubuntu and had python 3.7 install. What worked for me was, when ran
`ldconfig -p | grep libboost_python`

I noticed that the boost library ended with `_python37` so that is what I used.

```
	libboost_python37.so.1.67.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libboost_python37.so.1.67.0
	libboost_python37.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libboost_python37.so
	libboost_python27.so.1.67.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libboost_python27.so.1.67.0
	libboost_python27.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libboost_python27.so
```

So I ran 

```
sed -ie 's/boost_python-py34/boost_python37/' setup.py
pip3 install .
```

Then followed up with `pip3 install pybluez\[ble\]`

closed 2019-08-20 by stale[bot]

vladyslav2, 2020-12-01:
For everyone who still use an issue
```
sudo apt-get install libbluetooth-dev
sudo apt-get install libboost-python-dev
sudo apt-get install libboost-all-dev
git clone https://github.com/pybluez/pybluez
cd pybluez
pip3 install .
```