pybluez/pybluez #174: Can't Bind to RFCOMM Socket On Mac OS X

UsersDoProgram, 2017-12-21:
When binding to an RFCOMM socket in Python 3 on Mac OSX with PyBluez I get an error from the following code, 
`sock.bind((bluetooth.read_local_bdaddr()[0].decode,0))`
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/lightblue/_bluetoothsockets.py", line 60, in _getavailableport
    result, channelID, servicerecordhandle = BBServiceAdvertiser.addRFCOMMServiceDictionary_withName_UUID_channelID_serviceRecordHandle_(BBServiceAdvertiser.serialPortProfileDictionary(), "DummyService", None, None, None)
TypeError: 'int' object is not iterable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Volumes/RUSB/blueUtil.py", line 25, in test
    sock.bind(self.addr)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/bluetooth/osx.py", line 110, in bind
    return self._sock.bind(addrport)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/lightblue/_bluetoothsockets.py", line 297, in bind
    address = (address[0], _getavailableport(self.__conn.proto))
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/lightblue/_bluetoothsockets.py", line 62, in _getavailableport
    result, channelID, servicerecordhandle = BBServiceAdvertiser.addRFCOMMServiceDictionary_withName_UUID_channelID_serviceRecordHandle_(BBServiceAdvertiser.serialPortProfileDictionary(), "DummyService", None)
TypeError: Need 5 arguments, got 3

rgov, 2017-12-21:
I just merged in significant changes to support on macOS yesterday. What version are you using?

UsersDoProgram, 2017-12-21:
@rgov I'm on Mac OSX High Sierra with PyBluez 0.22.

rgov, 2017-12-21:
Can you try installing with

pip install --upgrade git+https://github.com/karulis/pybluez.git

to use the latest version? We haven’t released it to PyPI yet.

armandsm, 2018-01-02:
Hey, still having the same issues after installing the newest version of PyBluez onto MacOS High Sierra. Error code:
```
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/lightblue/_bluetoothsockets.py", line 60, in _getavailableport
    result, channelID, servicerecordhandle = BBServiceAdvertiser.addRFCOMMServiceDictionary_withName_UUID_channelID_serviceRecordHandle_(BBServiceAdvertiser.serialPortProfileDictionary(), "DummyService", None, None, None)
TypeError: 'int' object is not iterable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/armand/Desktop/pybluez_receivecode.py", line 12, in <module>
    server_sock.bind(("",port))
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/bluetooth/osx.py", line 110, in bind
    return self._sock.bind(addrport)
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/lightblue/_bluetoothsockets.py", line 297, in bind
    address = (address[0], _getavailableport(self.__conn.proto))
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/lightblue/_bluetoothsockets.py", line 62, in _getavailableport
    result, channelID, servicerecordhandle = BBServiceAdvertiser.addRFCOMMServiceDictionary_withName_UUID_channelID_serviceRecordHandle_(BBServiceAdvertiser.serialPortProfileDictionary(), "DummyService", None)
TypeError: Need 5 arguments, got 3
```

rgov, 2018-01-02:
Both reporters are using Python 3.x. Can you try Python 2.7, to see if it's a 3.x incompatibility or something else?

UsersDoProgram, 2018-01-02:
@rgov I'll try it now.

UsersDoProgram, 2018-01-02:
@rgov  When installing PyBluez on Python 2.7 I get the following error, 
Exception:
Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/Library/Python/2.7/site-packages/pip/commands/install.py", line 342, in run
    prefix=options.prefix_path,
  File "/Library/Python/2.7/site-packages/pip/req/req_set.py", line 778, in install
    requirement.uninstall(auto_confirm=True)
  File "/Library/Python/2.7/site-packages/pip/req/req_install.py", line 754, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/Library/Python/2.7/site-packages/pip/req/req_uninstall.py", line 115, in remove
    renames(path, new_path)
  File "/Library/Python/2.7/site-packages/pip/utils/__init__.py", line 267, in renames
    shutil.move(old, new)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 299, in move
    copytree(src, real_dst, symlinks=True)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 208, in copytree
    raise Error, errors
Error: [('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/NibClassBuilder.pyc', '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/NibClassBuilder.pyc', "[Errno 1] Operation not permitted: '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/NibClassBuilder.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/AppCategories.py', '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/AppCategories.py', "[Errno 1] Operation not permitted: '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/AppCategories.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/TestSupport.pyc', '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/TestSupport.pyc', "[Errno 1] Operation not permitted: '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/TestSupport.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/Signals.py', '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/Signals.py', "[Errno 1] Operation not permitted: '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/Signals.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/FndCategories.pyc', '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/FndCategories.pyc', "[Errno 1] Operation not permitted: '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/FndCategories.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/Signals.pyc', '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/Signals.pyc', "[Errno 1] Operation not permitted: '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/Signals.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/Conversion.pyc', '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/Conversion.pyc', "[Errno 1] Operation not permitted: '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/Conversion.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/NibClassBuilder.py', '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/NibClassBuilder.py', "[Errno 1] Operation not permitted: '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/NibClassBuilder.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools/TestSupport.py', '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Vers [... cut ...] bjC/PyObjCTools/AppHelper.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools', '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools', "[Errno 1] Operation not permitted: '/var/folders/6m/mxx8756x53v7p_vxmsq1t78m0000gn/T/pip-FmZU42-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC/PyObjCTools'")]

FynnMazurkiewicz, 2018-01-06:
This might just be a typo, but it should be `decode() ` instead of `decode`.

rgov, 2018-01-06:
If it works with the typo corrected, please submit a pull request.

FynnMazurkiewicz, 2018-01-07:
Not sure if you mean me, but I don't have an apple device available. Otherwise I totally would.

zbot473, 2018-08-28:
This is still broken. I tried what Fynn said, but it doesn't fix anything.

rgov, 2018-08-29:
It is still broken because nothing has changed. If you have the time to investigate, maybe you can help us get closer to the solution.

tomasero, 2018-09-07:
Any news on a fix for this?

rgov, 2019-07-22:
Keeping this issue open since I don't think it is resolved.

diogo-p-nunes, 2019-11-04:
Is the issue close to being solved? I just printed the returned value by the method BBServiceAdvertiser.addRFCOMMServiceDictionary_withName_UUID_channelID_serviceRecordHandle_(BBServiceAdvertiser.serialPortProfileDictionary(), "DummyService", None, None, None) and it outputs a negative integer (-536870212)

Peanutsmasher, 2019-12-18:
I can confirm that the issue is not yet resolved.

I tested the examples/simple/rfcomm-server.py with Python 3.7.5 on  OS X 10.15.2, resulting in "TypeError: Need 5 arguments, got 3".

rgov, 2019-12-18:
Not super useful info: -536870212 is 0xe00002bc which is a general kIOReturnError.

zellerman, 2020-03-15:
still happening...

tigerking, 2020-04-08:
still not work on Mac Catalina 10.15.2

tigerking, 2020-04-23:
fixed in my repo  with 2 commits @rgov  
https://github.com/tigerking/pybluez/commits/master

zellerman, 2020-04-23:
@tigerking I've tried your fork after `python setup.py install`-ing it in a clear venv and now I'm getting another backtrace. Is this LightAqueBlue supposed to be installed as well?
```
Traceback (most recent call last):
  File "/Users/zeller/PycharmProjects/timesheet/seggek.py", line 1, in <module>
    import bluetooth
  File "/Users/zeller/work/local/virtualenvs/scratch/lib/python3.7/site-packages/PyBluez-0.30-py3.7.egg/bluetooth/__init__.py", line 38, in <module>
    from .macos import *
  File "/Users/zeller/work/local/virtualenvs/scratch/lib/python3.7/site-packages/PyBluez-0.30-py3.7.egg/bluetooth/macos.py", line 1, in <module>
    import lightblue
  File "/Users/zeller/work/local/virtualenvs/scratch/lib/python3.7/site-packages/PyBluez-0.30-py3.7.egg/lightblue/__init__.py", line 160, in <module>
    from ._lightblue import *
  File "/Users/zeller/work/local/virtualenvs/scratch/lib/python3.7/site-packages/PyBluez-0.30-py3.7.egg/lightblue/_lightblue.py", line 29, in <module>
    from . import _LightAquaBlue
  File "/Users/zeller/work/local/virtualenvs/scratch/lib/python3.7/site-packages/PyBluez-0.30-py3.7.egg/lightblue/_LightAquaBlue.py", line 33, in <module>
    _FRAMEWORK_PATH)
ImportError: Cannot load LightAquaBlue framework, not found at /Users/zeller/work/local/virtualenvs/scratch/lib/python3.7/site-packages/PyBluez-0.30-py3.7.egg/lightblue/LightAquaBlue.framework

```

tigerking, 2020-04-23:
> @tigerking I've tried your fork after `python setup.py install`-ing it in a clear venv and now I'm getting another backtrace. Is this LightAqueBlue supposed to be installed as well? The 5 year old version from sourceforge no longer works...
> 
> ```
> Traceback (most recent call last):
>   File "/Users/zeller/PycharmProjects/timesheet/seggek.py", line 1, in <module>
>     import bluetooth
>   File "/Users/zeller/work/local/virtualenvs/scratch/lib/python3.7/site-packages/PyBluez-0.30-py3.7.egg/bluetooth/__init__.py", line 38, in <module>
>     from .macos import *
>   File "/Users/zeller/work/local/virtualenvs/scratch/lib/python3.7/site-packages/PyBluez-0.30-py3.7.egg/bluetooth/macos.py", line 1, in <module>
>     import lightblue
>   File "/Users/zeller/work/local/virtualenvs/scratch/lib/python3.7/site-packages/PyBluez-0.30-py3.7.egg/lightblue/__init__.py", line 160, in <module>
>     from ._lightblue import *
>   File "/Users/zeller/work/local/virtualenvs/scratch/lib/python3.7/site-packages/PyBluez-0.30-py3.7.egg/lightblue/_lightblue.py", line 29, in <module>
>     from . import _LightAquaBlue
>   File "/Users/zeller/work/local/virtualenvs/scratch/lib/python3.7/site-packages/PyBluez-0.30-py3.7.egg/lightblue/_LightAquaBlue.py", line 33, in <module>
>     _FRAMEWORK_PATH)
> ImportError: Cannot load LightAquaBlue framework, not found at /Users/zeller/work/local/virtualenvs/scratch/lib/python3.7/site-packages/PyBluez-0.30-py3.7.egg/lightblue/LightAquaBlue.framework
> ```

above issue root cause is missing symbol link for LightAquaBlue.framework,  you can either create symbol link manually 
Or try install from whl file to see if it can solve your issue.  I put the whl file at 
https://github.com/tigerking/pybluez/releases

zellerman, 2020-04-23:
@tigerking ok, I managed it. I noticed that you have to bdist_wheel  it. Didn't work from virtualenv for some reason but after building it in the system python environment, I was able to install the package and IT WORKS! 🦩  Anyway, the wheel name possibly depends on the machine you build. Mine was `PyBluez-0.30-cp37-cp37m-macosx_10_`**14**`_x86_64.whl` not `PyBluez-0.30-cp37-cp37m-macosx_10_`**13**`_x86_64.whl`
`

niemyjski, 2020-05-01:
I'm also running into this on macOS using requirements.txt and venv

miamilabs, 2021-03-04:
Is there a way that to make this work on OSX 10.15?
Kinda  i am not python developer but playing around.

cgalpin, 2021-03-10:
I have the same issue on OS X 11.2. IF anyone can help get pybluez installed and working I'd appreciate it. I get what looks like a successful install, but when I use it i get

```
    import bluetooth._bluetooth as bluez
  File "/usr/local/lib/python3.8/site-packages/PyBluez-0.30-py3.8.egg/bluetooth/__init__.py", line 38, in <module>
    from .macos import *
  File "/usr/local/lib/python3.8/site-packages/PyBluez-0.30-py3.8.egg/bluetooth/macos.py", line 1, in <module>
    import lightblue
  File "/usr/local/lib/python3.8/site-packages/lightblue/__init__.py", line 160, in <module>
    from _lightblue import *
ModuleNotFoundError: No module named '_lightblue'
```

tia, charles

viral-sangani, 2021-04-26:
I am also getting issues while installing pybluez on OS X 11.2. Any help would be appreciated.

SecretA, 2024-01-05:
To Install PyBluez you must use an old Python version, I think the last supported version that works is Python 3.7.9, I tried it on Python 3.8 it doesn't work.