pybluez/pybluez #175: Error on Windows connecting to a RFCOMM Socket

UsersDoProgram, 2017-12-21:
I created a server that binds to the address  ('',0) in pybluez using the RFCOMM protocol, the code is here:
`from bluetooth import *`
`sock = BluetoothSocket(RFCOMM)`
`sock.bind(('',0))`
`sock.listen(1)`
`c,addr = s.accept()`
`c.send("Hey")`
`print(c.recv(1024))` 
Then a client, 
`from bluetooth import *`
`client = BluetoothSocket(RFCOMM)`
`client.connect(('',0))`
`client.send("Hi")`
`print(client.recv(1024))`
The client.connect line is the line that raises the error, 
OSError: The requested address is not valid in context

What can I do to fix this?

FynnMazurkiewicz, 2017-12-23:
Since you are using an `RFCOMM`-type socket, you should provide a `(host, port)` pair to `client.connect`. (As specified in the docs, see the folder `/docs`.) 
But you're providing an empty host.
This should be the bluetooth mac-address of the server instead.

Sidenote: I can't run code right now and haven't used pyBluez in a bit of time, just thought I'd come back and answer a few issues. Merry Christmas!

UsersDoProgram, 2018-01-02:
@FynnMazurkiewicz Thanks, I'll try this now hopefully I have some luck. I am a little late but Happy New Year!

closed 2019-07-30 by stale[bot]