pybluez/pybluez #176: Raspberry Pi 3 BLE beaons

akshaypm, 2018-01-02:
I am trying to list out the iBeacons that are broadcasting on raspberry Pi 3 , I have installed pybluez and pygattlib as well. But when i run the example in example/BLE/ beacon_scan.py its returns nothing and only prints "DONE" but the scan.py scripts prints the beacon name and UUID's

Please help me

rgov, 2018-01-02:
You're using beacon_scan.py on the Raspberry Pi 3? What version of Python, and what version of pybluez did you install? Can you try installing it from the GitHub sources?

If the scan.py script works, why not use it instead?

akshaypm, 2018-01-03:
yes i am using the onboard BLE radio on Raspberry Pi, i am using python 2.7 i downloaded Source files of bluez and installed manually as of now bluez is on 5.48.

If i use scan.py py its not returning UUID and also no API documentation for beacons so that i will be able to get the RSSI of the scanned beacons but when i use hcitool lescan it returns UUID, RSSI,MAC Address but not through code
Do i have to change any configuration on beacons

ukBaz, 2018-01-04:
> My basic doubt is does latest version of Raspberry pi i.e. Raspberry pi 3 inbuilt ble 4.0 radio enough for running beacon scanner or does it require a external USB BLE dongle to be connected to Pi3

It sounds like by using `hcitool` that you have proved it is not the hardware.

I had a quick look through the pygattlib BeaconService and it appears to do various checks to only report iBeacons. This suggests to me that your iBeacons aren't in the format that beacon_scan.py is expecting.

Do you have a second Raspberry PI? Could you run example/BLE/beacon.py on one and see if beacon_scan.py can find that beacon on the other one?

Another option might be to run `sudo btmon` in a different terminal as your run beacon_scay.py. btmon will report Advertisement events. If you report the values back here maybe one of the developers can tell if your beacon has the correct values.

PikWay, 2018-01-12:
Hi, 
I have the same issue on Pi Zero W. 
In `hcitool lescan` I can see beacon:

D0:39:72:A0:34:57 (unknown)
> HCI Event: LE Meta Event (0x3e) plen 39                                                                                                                                          #35 [hci0] 201.185798
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scannable undirected - ADV_SCAN_IND (0x02)
        Address type: Public (0x00)
        Address: D0:39:72:A0:34:57 (Texas Instruments)
        Data length: 27
        Company: Apple, Inc. (76)
          Type: iBeacon (2)
          UUID: fe78b237-5282-f19f-374f-0706e4a1dca2
          Version: 23810.11571
          TX power: -59 dB
        RSSI: -69 dBm (0xbb)
D0:39:72:A0:34:57 (unknown)
> HCI Event: LE Meta Event (0x3e) plen 12                                                                                                                                          #36 [hci0] 205.209500
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scan response - SCAN_RSP (0x04)
        Address type: Public (0x00)
        Address: D0:39:72:A0:34:57 (Texas Instruments)
        Data length: 0
        RSSI: -69 dBm (0xbb)

But beacon_scan cant see it: 

pi@raspitestbed:~ $ sudo python3 testBLE.py
@ RAW Open: python3 (privileged) version 2.22                                                                                                                                        {0x0004} 556.642757
@ RAW Close: python3                                                                                                                                                                 {0x0004} 556.644095
@ RAW Open: python3 (privileged) version 2.22                                                                                                                                 {0x0004} [hci0] 556.645103
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                                                                                                         #39 [hci0] 556.646231
        Type: Active (0x01)
        Interval: 10.000 msec (0x0010)
        Window: 10.000 msec (0x0010)
        Own address type: Public (0x00)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                        #40 [hci0] 556.646872
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                                                                                             #41 [hci0] 556.654846
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                        #42 [hci0] 556.655349
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                                                                                             #43 [hci0] 556.657088
        Scanning: Disabled (0x00)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                        #44 [hci0] 556.658390
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
{}
Done.
@ RAW Close: python3                                                                                                                                                          {0x0004} [hci0] 556.701187

closed 2019-07-30 by stale[bot]