pybluez/pybluez #180: Windows 10 installation error(Python: 3.6.4)

Vikhyatcis, 2018-01-09:
I run pip on cmd(admin). 
It gives this error:
C:\Python\Scripts>call pip install pybluez
Collecting pybluez
  Using cached PyBluez-0.22.zip
    Complete output from command python setup.py egg_info:
    Could not find the Windows Platform SDK

Command "python setup.py egg_info" failed with error code 1 in C:\Users\VKSN\AppData\Local\Temp\pip-build-v6w6gkhr\pybluez\

I have windows 10 and python 3.6.4.

nickums, 2018-02-16:
I have the exactly the same error on windows 10, immediaiely after installing the latest windows SDK

IC088, 2018-03-06:
Help, same issue with the rest

mitjad, 2018-05-28:
me2

nchauhan5, 2018-05-29:
Any update on this issue ? Facing the same issue myself. Please help

rgov, 2018-05-29:
As far as I know, no one is contributing to Windows support in PyBluez right now.

wonhyeongseo, 2018-05-30:
So a user must install Linux to use PyBluez for a project? If so, what fits best, Ubuntu?

karulis, 2018-05-30:
You can use with no problem older version of interpreter and binaries here:
https://www.lfd.uci.edu/%7Egohlke/pythonlibs/#pybluez
I guess there must be some problem with W10 and Python3.6, but I dont have setup to verify it

karulis, 2018-05-30:
Could You run pip with -v option:
pip -v install pybluez

kennyjudo, 2018-06-11:
Hi karulis, I have already use the "pip -v install pybluez", but I am still facing the same error 
"pip._internal.exceptions.InstallationError: Command "python setup.py egg_info" failed with error code 1 in C:\Users\KENNYC~1\AppData\Local\Temp\pip-install-taev_jtj\pybluez\"

karulis, 2018-06-11:
Just to clarify things, You guys have build requirements installed for windows:
Visual C++ 2010 Express for build for Python 3.3 or newer
Thought it seems it moght be bit different for windows 10, other packages have similar issues
https://github.com/pygraphviz/pygraphviz/issues/121

demanbart, 2018-06-15:
I've had the same problem with windows 10, python 3.6 (32bit).
When manually installing I've had no issues, so if You need this you can do run the following in a cmd window.
```
git clone https://github.com/pybluez/pybluez
cd pybluez
python setup.py install

```

This gave me no errors and I was able to run the _simple inquire example_ without issues.
My guess is that there is a difference between the code on Github and the version available via pip.

kennyjudo, 2018-06-17:
Hi demanbart,

Thanks  for your help, but it pop up another error " Could not find the
Windows Platform SDK" and I still can't install, i am using win10, 64bit,
anaconda, python 3.6, spyder 3.2.8

kennyjudo, 2018-06-17:
Hi all, I can install the pybluez after I install the win10 SDK and Visual
C++ build tools.

thanks

demanbart, 2018-06-19:
I just noticed the find_MS_SDK function in the setup.py script. As You point out an SDK has to be installed. Following paths were checked when running the setup.py script:
C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A
C:\Program Files (x86)\Microsoft Platform SDK for Windows XP
C:\Program Files (x86)\Microsoft Platform SDK
C:\Program Files\Microsoft SDKs\Windows\v10.0A
C:\Program Files\Microsoft Platform SDK for Windows XP
C:\Program Files\Microsoft Platform SDK
C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A
C:\Program Files (x86)\Microsoft Platform SDK for Windows XP
C:\Program Files (x86)\Microsoft Platform SDK

If one of these is found, the installation proceeds, otherwise it is aborted with error: "Could not find the Windows Platform SDK" message. It would be elegant to point out the need for the specific SDK in the message.

krushiraj, 2018-06-28:
I had problem which says `LINK : fatal error LNK1158: cannot run 'rc.exe'
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\link.exe' failed with exit status 1158
`.  It can be solved by copying `rc.exe, rcdll.dll` files from `C:\Program Files (x86)\Windows Kits\8.1\bin\x86` to `C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin`.

nickums, 2018-06-29:
Thanks Demanbart for the lucid solution and explanation of the install rewliance on SDK.
Having sucessfully cloned and  run, python setup.py install,  on trying the simple example
I get ModuleNotFoundError: No module named 'bluetooth'.
Must be something simple but I'm flummoxed.

nickums, 2018-06-29:
I'm not so lucky.  Nothing from bluez  in site packages.  Must look into that setup.py ....

nickums, 2018-07-02:
No such luck. 
I've copied the bluetooth directory into site_packages  and can see _init_  failing with
No module named 'bluetooth._msbt' although _msbt.py exists
I obviouslyneed help understanding enough about modules

nickums, 2018-07-02:
had another go at regular install throws up a pybluez problem with pip:
------------------------------------------------------------------------------------------------------------
C:\Python\Python36>pip install pybluez
Requirement already satisfied: pybluez in c:\users\nick\appdata\local\programs\python\python36-32\lib\site-packages\pybluez-0.22-py3.6-win32.egg
You are using pip version 9.0.1, however version 10.0.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

C:\Python\Python36>python -m pip install --upgrade pip
Requirement already up-to-date: pip in c:\python\python36\lib\site-packages (10.0.1)

nickums, 2018-07-08:
ah I begin to see - it went to the 32bit directory I had forgotten about. On removing all old Python stuff and trying again:
:\Python\Python36\GitHub\pybluez>python setup.py install
running install
running bdist_egg
running egg_info
writing PyBluez.egg-info\PKG-INFO
writing dependency_links to PyBluez.egg-info\dependency_links.txt
writing eager_resources to PyBluez.egg-info\eager_resources.txt
writing requirements to PyBluez.egg-info\requires.txt
writing top-level names to PyBluez.egg-info\top_level.txt
error: package directory 'bluetooth' does not exist

Skyr3ndWolf, 2018-07-24:
Guys, I tried installing by cloning the repository from GitHub. When i cd into the pybluez directory and run the command "python setup.py install", it says 'python' is not recognized as an internal or external command,operable program or batch file. I'm using python 3.7. Please help

c-w, 2018-07-26:
Here are Windows 10 installation steps that worked for me. The steps are assuming you're using a conda-based Python installation such as [Anaconda](https://www.anaconda.com/download/).

Step 1: Download the latest code

```
git clone https://github.com/pybluez/pybluez
cd pybluez
```

Step 2: Create an environment for pybluez with your desired Python version. The example below uses Python 3.6 but the same steps will also work for other versions such as Python 2.7.

```
conda create -n pybluez python==3.6
activate pybluez
```

Step 3: Build the library. Note that you must have installed the correct C++ compiler for your desired Python version before executing the commands. You can find download links to the compilers on the Python wiki [WindowsCompilers page](https://wiki.python.org/moin/WindowsCompilers).

```
python setup.py install
python setup.py bdist_wheel
```

These two steps will create a `.whl` file in the `dist` folder. You can now use pybluez in another project by pip-installing this `.whl` file into your project's virtualenv.

Skyr3ndWolf, 2018-07-27:
@c-w Thank you!! i Don't run Anaconda, but you Pybluez wheel was of help.

skjerns, 2018-10-09:
> I've also made available 64-bit wheels based on pybluez commit [23d3471](https://github.com/pybluez/pybluez/commit/23d34718046b2ad40bd519bea26cac38d33933a7) for the following Python versions:
> 
> * [Python 3.7](https://1drv.ms/u/s!AtLn8ELpA_G9gQDn_DXUSjX8-jgD)

thank you so much

murphychentw, 2018-12-18:
I successfully installed pybluez for Windows 10 with Python 3.7.

1. Download and run "Visual Studio Installer": https://www.visualstudio.com/pl/thank-you-downloading-visual-studio/?sku=BuildTools&rel=15
2. Install "Visual Studio Build Tools 2017", check "Visual C++ build tools" and "Universal Windows Platform build tools"
3. git clone https://github.com/pybluez/pybluez
4. cd pybluez
5. python setup.py install

nickums, 2019-01-26:
Having downloaded the suggested  PyBluez-0.22-cp37-cp37m-win_amd64.whlldircetory and using up to date pip 19.0.1,I was hoping  the followin gsimple insatll for a ,whl  would work;
I can't understand why it does not.

C:\Python\Python36>pip install PyBluez-0.22-cp37-cp37m-win_amd64.whl 
 Requirement 'PyBluez-0.22-cp37-cp37m-win_amd64.whl' looks like a filename, but the file does not 
 exist
PyBluez-0.22-cp37-cp37m-win_amd64.whl is not a supported wheel on this platform.

The specified .whl does exist :-

C:\Python\Python36>dir *.whl
   Directory of C:\Python\Python36
   25/01/2019  14:27    <DIR>          PyBluez-0.22-cp37-cp37m-win_amd64.whl
   etc

rgov, 2019-01-27:
As with any open source project, if you are reporting a separate problem than the one described, please file a new issue.

You seem to be trying to install a wheel for Python 3.7 from a directory that indicates that you are using Python 3.6. Please make sure you are using matching Python versions.

francosax, 2019-03-07:
I've solved this problem (for a python ver. 2.7.9 and win10) by : 

1 - installing the MS SDK at : https://developer.microsoft.com/it-it/windows/downloads/windows-10-sdk

2 - and chose, during SDK installation phase, one of the path that *demanbart* suggested, eg:  
  C:\Program Files (x86)\Microsoft Platform SDK

It worked fine for me.

NihilisticRealist, 2019-05-20:
https://pypi.org/project/PyBluez-win10/ Worked for me.

beadysea, 2019-07-25:
Here are the steps I took without a conda environment.
## Step 1: Install or upgrade python setuptools
If you're not sure setuptools is installed then enter `pip list` in a cmd terminal.

In cmd terminal type:
`pip install setuptools` to install setuptools or `pip install --upgrade setuptools` to make sure you have the latest version.

## Step 2: Make sure you have the required C\C++compiler.
As I discovered through trial and error, you can't just use any C\C++ compiler it has to be a specific one based on the version of python you're using. [Cannon 2018] covers this in detail and gives comprehensive install instructions.  

A quick check with `python -VV`gave me:
```
Python 3.7.4  (tags/v3.7.4:e09359112e, Jul  8 2019, 20:34:20) [MSC v.1916 64 bit (AMD64)] on win32
```
According to [Cannon 2018] I need Microsoft Visual C++ 14.0 (standalone or Visual Studio 2017) and I'm going to need the x64 version.

I chose to install Visual Studio 2017. With everything setup I tried `pip install pybluez` and it failed "Could not find the Windows Platform SDK" 

Here's the gotcha! The good folks at Microsoft decided to let you choose which tools to install so the default install of Visual Studio does not install the build tools we need. As @murphychentw states [here](https://github.com/pybluez/pybluez/issues/180#issuecomment-448102727) Desktop Development with C++ and Universal Windows Platform Development need to be checked during install.  This can be done after installation by going to Tools >> Get Tools and Features.. inside Visual Studio.

**Important**
The Universal Windows Platform Development tools installs the Windows 10 SDK needed to build Windows 10 applications.

If you're not using Visual Studio follow the link below to download the SDK.
[Download Windows 10 SDK](https://developer.microsoft.com/en-us/windows/downloads/windows-10-sdk). Thanks to @francosax for the [insight here](https://github.com/pybluez/pybluez/issues/180#issuecomment-470457218)

## Step 3: Install PyBluez 

In a cmd terminal I typed:

  `pip install pybluez`

Great! PyBluez v 0.22 successfully installed. 

## Step 4: Test everything works

My first effort to import bluetooth failed with ModuleNotFoundError: No module named 'bluetooth._msbt'. Similar to @nickums [experience](https://github.com/pybluez/pybluez/issues/180#issuecomment-401310129) . 
All I did was close my cmd terminal, started a new one and typed python then:
```
`>>> import bluetooth as bt
>>> print(bt.discover_devices())
['30:21:3C:C9:19:95', '22:22:22:EA:3F:5B', '00:BA:55:56:F1:09', '04:C2:3E:05:A3:30']
```

[Cannon 2018]: https://wiki.python.org/moin/WindowsCompilers

oana-oana, 2019-08-23:
i did not manage to get rid of this "ModuleNotFoundError: No module named 'bluetooth._msbt'.". i closed the terminal and started a new one, the same. I even performed a reset and nothing changed.
i tried both import bluetooth as bt and trying to run inquiry example and i get the same error. Am i missing something?

Gitarzanee, 2019-09-23:
Nickums, had a good clue, when he posted all the paths that were searched looking for the SDK.
I think microsoft has changed the path naming, in later VS drops, so that is why setup.py does not find the SDK.
So I found the actual path of the SDK and modified setup.py to find it.
Ran python setup.py install
FIXED.  Then I did not have the No Windows Platform SDK error issue.

But running,  now I have the No module bluetooth.msbt error issue.

I may give up.  Too many variables, project seems unsupported.  May move to Java instead.

nickums, 2019-09-25:
Hi.
I managed to get pybluez to work on my Python3.6
after finding the right wheel file:
pip install Pybluez-0.22.cp36-cp36m-win_amd64.whl

Good luck!
Nick

julianhapke, 2019-11-01:
Another Solution for Visual Studio 19:

`git clone https://github.com/pybluez/pybluez`

edit the `setup.py`
under >=3.5 (Line 55) just add the install path of your Visual Studio Install, e.g.

`candidate_paths.append(r'Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105')  # Visual Studio 14.2`

now
`python setup.py install`
should work.

steam3d, 2019-11-24:
i try everything, but i get this 
` LINK : fatal error LNK1158: не удается запустить "rc.exe"
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\link.exe' failed with exit status 1158`

rgov, 2019-11-24:
Please file a new issue and include more details about your system and the
complete build log. Thank you.

jonnekendy, 2019-12-19:
https://errorslab.com

rgov, 2019-12-28:
Please try to install PyBluez 0.23 from PyPI and see if it works on Windows.

niladri30, 2020-01-17:
I have tried with `pip install PyBluez-win10 `, which worked for me.

rgov, 2020-01-17:
PyBluez-win10 is unofficial and was released over a year ago. PyBluez 0.23 has been released and attempted to fix the Windows 10 issue. If it does not work, please let us know so we can improve it. I do not recommend using PyBluez-win10.

svtesan, 2020-04-06:
> msbt

I got the same error and it was because I was running python from the pybluez dir. I cd'd to an outside dir, and voila it worked! If you run python from within the pybluez dir, it tries to load the files _from_ the dir and not the wheel. 

Steps:
1. Clone git repo
2. open conda prompt (or any other cmd prompt with python in path)
3. CD to the cloned pybluez dir
4.  python setup.py bdist_wheel
5. pip install dist\PyBluez*.whl 
6. IMPORTANT: cd to a dir outside of pybluez dir
7. Now verify with: 
python
>> import bluetooth as bt
>>bt.discover_devices()

koog1000, 2020-04-18:
@svtesan Thanks for posting this! I was getting pretty frustrated trying to figure why the build wasn't working.

laksh225, 2020-05-22:
I guess it's time to close this issue as the problems have been solved.

closed 2020-06-25 by Vikhyat2603

saurabhpant93, 2020-11-29:
> Here are Windows 10 installation steps that worked for me. The steps are assuming a conda-based Python installation such as [Anaconda](https://www.anaconda.com/download/).
> 
> Step 1: Download the latest code
> 
> ```
> C:\Users\me> git clone https://github.com/pybluez/pybluez
> C:\Users\me> cd pybluez
> ```
> 
> Step 2: Create an environment to build pybluez with the desired Python version. The example below uses Python 3.6 but the same steps will also work for other versions such as Python 2.7, etc.
> 
> ```
> C:\Users\me\pybluez> conda create -n pybluez python==3.6
> C:\Users\me\pybluez> activate pybluez
> ```
> 
> Step 3: Build the library. Note that the correct C++ compiler for the desired Python version must be installed before running the build. The download links to the compilers can be found on the Python wiki [WindowsCompilers page](https://wiki.python.org/moin/WindowsCompilers).
> 
> ```
> (pybluez) C:\Users\me\pybluez> python setup.py install
> (pybluez) C:\Users\me\pybluez> python setup.py bdist_wheel
> (pybluez) C:\Users\me\pybluez> deactivate
> ```
> 
> These two steps will create a `.whl` file in the `dist` folder. The pybluez library can now be used in another project by pip-installing this wheel file into the project's virtualenv. For example:
> 
> ```
> C:\Users\me\pybluez> cd ..\project
> C:\Users\me\project> conda create -n project python==3.6
> C:\Users\me\project> activate project
> (project) C:\Users\me\project> pip install ..\pybluez\dist\PyBluez-0.22-cp36-cp36m-win_amd64.whl
> (project) C:\Users\me\project> python -c "import bluetooth; print(bluetooth.__version__)"
> ```
> 
> I've also made available 64-bit wheels based on pybluez commit [23d3471](https://github.com/pybluez/pybluez/commit/23d34718046b2ad40bd519bea26cac38d33933a7) for the following Python versions:
> 
> * [Python 2.7](https://1drv.ms/u/s!AtLn8ELpA_G9frRzQuMKJ1QixKw)
> * [Python 3.5](https://1drv.ms/u/s!AtLn8ELpA_G9gQHhYULrCIzs-u8A)
> * [Python 3.6](https://1drv.ms/u/s!AtLn8ELpA_G9fyHIyPR-5uvY6d8)
> * [Python 3.7](https://1drv.ms/u/s!AtLn8ELpA_G9gQDn_DXUSjX8-jgD)

Thanks @c-w ! I tried the steps for Python 3.9 and it worked (without conda). Here are some other tips for Python 3.9 on Windows 10-
1) You need wheels installed- pip install wheel before python setup.py bdist_wheel 
2) Must follow [this ](https://wiki.python.org/moin/WindowsCompilers)for installation of Visual C++ compiler, especially- "In Build tools, install C++ build tools and ensure the latest versions of MSVCv142 - VS 2019 C++ x64/x86 build tools and Windows 10 SDK are checked." part.
3) Careful pip vs pip3
4) I got this error-ModuleNotFoundError: No module named 'bluetooth._msbt', resolved by just restarting the terminal

Thanks,
Saurabh