pybluez/pybluez #181: Installing PyBluez on Mac OSX 10.13.1

Max604, 2018-01-15:
Hello,

In my IDE (PyCharm) I am receiving this message for `import bluetooth`: 'No module named bluetooth'.

How can I troubleshoot this problem? PyBluez installed fine after installing Pyobjc.

Thanks,
Max

rgov, 2018-01-15:
Can you try installing from the latest git repo? `pip install --upgrade git+https://github.com/rgov/pybluez.git#egg=pybluez`

Max604, 2018-01-16:
It ran successfully but I still can't import `bluetooth`. How can I make it verbose and find the error?

rgov, 2018-01-16:
Oops, I updated the command above for you to run one more time: `pip install --upgrade git+https://github.com/karulis/pybluez.git#egg=pybluez`

Max604, 2018-01-16:
My output for the following:
pip install --upgrade git+https://github.com/karulis/pybluez.git#egg=pybluez
Collecting pybluez from git+https://github.com/karulis/pybluez.git#egg=pybluez
  Cloning https://github.com/karulis/pybluez.git to /private/var/folders/b6/tklyd_cx07b3cyzr93vg_1840000gn/T/pip-build-3hx2h5/pybluez
Requirement already up-to-date: pyobjc-core>=3.1 in /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (from pybluez)
Requirement already up-to-date: pyobjc-framework-Cocoa>=3.1 in /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (from pybluez)
Installing collected packages: pybluez
  Found existing installation: PyBluez 0.22
    Uninstalling PyBluez-0.22:
      Successfully uninstalled PyBluez-0.22
  Running setup.py install for pybluez ... done
Successfully installed pybluez-0.22

I wouldn't be able to create a mac osx virtual environment and reinstall the pycharm ide to test. Isn't there other ways to find some error that says why it isn't doesn't exist when i try to import it in my ide?

Max604, 2018-01-17:
Bump, I really need to fix this issue to work on my research.

Max604, 2018-01-17:
Traceback (most recent call last):
  File "/Users/Archer/Desktop/Bluetooth.py", line 1, in <module>
    import bluetooth
ModuleNotFoundError: No module named 'bluetooth'

karulis, 2018-01-17:
Does this happen only with IDE or also from command line?

Max604, 2018-01-17:
I just tried this on PyCharm and IDLE

karulis, 2018-01-17:
Please try command line for me:)

karulis, 2018-01-17:
Thought in idle it should be same, 1 min

Max604, 2018-01-17:
All right, thank you so much for assisting me

Max604, 2018-01-17:
Command line also outputs the same result as IDLE, exactly the same

karulis, 2018-01-17:
ok can You paste output of(maybe as attached file)
sudo pip install --upgrade pybluez -v

Max604, 2018-01-17:
[verbose.txt](https://github.com/karulis/pybluez/files/1638752/verbose.txt)

Max604, 2018-01-17:
[verbose.txt](https://github.com/karulis/pybluez/files/1638928/verbose.txt)

Max604, 2018-01-18:
Any idea what to do @karulis

rgov, 2018-01-18:
Here's what I did, on macOS 10.13.2, to make sure it works:

```
$ virtualenv .venv
$ . .venv/bin/activate
(.venv) $ pip install --upgrade git+https://github.com/karulis/pybluez.git#egg=pybluez
...
(.venv) $ python
Python 2.7.10 (default, Jul 15 2017, 17:16:57) 
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.31)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import bluetooth
>>>
```

So I can confirm that it installs fine normally. It doesn't look like this is a problem with pybluez necessarily, maybe just a problem with your IDE.

After running those commands, the `bluetooth` package is installed here: `.venv/lib/python2.7/site-packages/bluetooth` and then this directory is set to be searched in the Python search path:

```
(.venv) $ python -c 'import sys; print(sys.path)'
['', ..., '.venv/lib/python2.7/site-packages']
```

So I would first figure out where the `bluetooth` package was installed when you ran `pip install` and then make sure that directory appears in `sys.path`. If it doesn't, you can append to that variable just like any other Python list.

Max604, 2018-01-19:
Yes, that all worked! However, it is installed in 2.7, is it possible to install it on python 3+ just by appending '3' to commands like pip (pip3), python (python3), etc... Thanks

Max604, 2018-01-19:
Thanks! Sorry if that was nooby, I've only just begun learning this stuff to do some cool stuff with my raspberry pi.

jaraco, 2018-02-24:
I suggest closing this issue.

closed 2018-02-24 by Max604