pybluez/pybluez #184: Running on Linux without "sudo"

madneon, 2018-01-22:
Is there a way to use `pybluez` without sudo?
Right now I'm getting error on `hci_send_cmd(...)` - `error: (1, 'Operation not permitted')`

rgov, 2018-01-22:
What's the backtrace?

I suspect (but do not know) that pybluez can do many things without root, but maybe things like changing the hardware address or device classification might require more privileges.

madneon, 2018-01-23:
for eg. from [https://github.com/switchdoclabs/iBeacon-Scanner-](https://github.com/switchdoclabs/iBeacon-Scanner-)

> $ python testblescan.py 
> ble thread started
> Traceback (most recent call last):
>   File "testblescan.py", line 19, in <module>
>     blescan.hci_enable_le_scan(sock)
>   File "/home/madneon/Desktop/bt2/iBeacon-Scanner-/blescan.py", line 79, in hci_enable_le_scan
>     hci_toggle_le_scan(sock, 0x01)
>   File "/home/madneon/Desktop/bt2/iBeacon-Scanner-/blescan.py", line 103, in hci_toggle_le_scan
>     bluez.hci_send_cmd(sock, OGF_LE_CTL, OCF_LE_SET_SCAN_ENABLE, cmd_pkt)
> _bluetooth.error: (1, 'Operation not permitted')

btashton, 2018-03-12:
@madneon 
In order to execute an LE scan with bluez you need to have extra capabilities.  You should be careful doing this on the system python binary since that will give any python process the ability to have these privileged capabilities.

If you create a virtualenv then you can do something like this:
`sudo setcap 'cap_net_raw,cap_net_admin+eip' venv/bin/python3.6`

RafalSkolasinski, 2018-11-20:
Check output of 
```
file venv/bin/python3
```

Most likely it is a symbolic link to `venv/bin/python3.7m` or similar.
Try executing command providing a path to python binary, not symlink.

closed 2019-07-30 by stale[bot]