pybluez/pybluez #185: Get RSSI in DiscoveryService

pedrofelipest, 2018-01-23:
How to get value of rssi in DiscoveryService?

Max604, 2018-01-25:
Bump, I am also looking for the answer for this

rgov, 2018-01-25:
I don't know the answer directly, but I searched the project for `rssi` and I found a few things that look like good starting points:

* [`examples/simple/asynchronous-inquiry.py`](https://github.com/karulis/pybluez/blob/a0b226a61b166e170d48539778525b31e47a4731/examples/simple/asynchronous-inquiry.py) is a much simpler example that shows that subclasses of `bluetooth.DeviceDiscoverer` have the `device_discovered()` method called with an RSSI parameter.
* [`examples/advanced/inquiry-with-rssi.py`](https://github.com/karulis/pybluez/blob/a0b226a61b166e170d48539778525b31e47a4731/examples/advanced/inquiry-with-rssi.py) has a promising hit on the filename.

The README mentions that both of these are Linux-only, however. Please report back if you get them to work on other platforms.

* [`examples/ble/beacon_scan.py`](https://github.com/karulis/pybluez/blob/a0b226a61b166e170d48539778525b31e47a4731/examples/ble/beacon_scan.py) also mentions extracting RSSI from BLE beacons. It also might be Linux only.

Please report back if you get it to work, so others can find this information.

Max604, 2018-01-26:
> examples/ble/beacon_scan.py
I believe this only works with beacons, so it doesn't work with detecting my other Bluetooth devices, and what I desire is to get the RSSI from a BLE device.

They don't work on OSX afaik because it has a different Bluetooth setup from Linux systems.

closed 2019-07-30 by stale[bot]