pybluez/pybluez #187: IOError: Only one usage of each socket address (protocol/network address/port) is normally permitted.

KarolLyz, 2018-02-06:
Hello,

I have a problem with closing connection. I got error like below:
IOError: Only one usage of each socket address (protocol/network address/port) is normally permitted.

Scenario:
Im connected from my laptop with Windows to 2 phones and below code is executrd fine with devAddr and Port of 1st phone.

def call_status(devAddr, Port, path):
    """block!"""
    #Call_answer
    Responses.fname_response(call_status.__name__, path)
    Responses.time_response("start", path)
    time.sleep(5)
    s = bluetooth.BluetoothSocket(bluetooth.RFCOMM)
    conn = s.connect((devAddr, Port))
    s.send("AT+CLCC\r")
    time.sleep(4)
    response=str(s.recv(1024))
    Responses.at_response(response, path)
    time.sleep(5)
    s.close

After that I execute one more time this function with 2nd phone (another devAddr but same port) and i have revive this error.

I suppose there is some problem with closing connection with previous phone. I also in many cases need to use time.sleep long enough to not this kind of error.
Can anybody help with this?

gobftald, 2018-02-06:
don't you miss parentheses after close?

s.close()

KarolLyz, 2018-02-07:
Sorry, i Copy without (). One more time code:
def call_status(devAddr, Port, path):
    Responses.fname_response(call_status.__name__, path)
    Responses.time_response("start", path)
    s = bluetooth.BluetoothSocket(bluetooth.RFCOMM)
    time.sleep(2)
    conn = s.connect((devAddr, Port))
    s.send("AT+CLCC\r")
    time.sleep(2)
    response=str(s.recv(1024))
    Responses.at_response(response, path)
    time.sleep(1)
    s.close()

KarolLyz, 2018-03-05:
Any idea how to handle with it?

zkycloudz, 2018-10-02:
same problem with me, Have you find a way to solve? I am really thankfull if you can share to me.
Btw, I am using python 2.7.15 and pybluez 0.22

closed 2019-07-30 by stale[bot]