pybluez/pybluez #189: Missing SDP data

matealex, 2018-02-08:
I have a strange problem when trying the samples for RFCOMM server on a Raspberry Pi
If i run the script with python3 I can detect the service from other computer, but if i run it with python2.7 i can't detect it. 

"sudo sdptool browse local" outputs this:
With python 2.7:

> 
> Service Name: SampleServer
> Service RecHandle: 0x1000d
> Protocol Descriptor List:
>   "L2CAP" (0x0100)
>   "RFCOMM" (0x0003)
>     Channel: 1
> Profile Descriptor List:
>   "Serial Port" (0x1101)
>     Version: 0x0100

With python 3+:

> Service Name: SampleServer
> Service RecHandle: 0x1000d
> Service Class ID List:
>   UUID 128: 34b1cf4d-1069-4ad6-89b6-e161d79be4d8
>   "Serial Port" (0x1101)
> Protocol Descriptor List:
>   "L2CAP" (0x0100)
>   "RFCOMM" (0x0003)
>     Channel: 1
> Profile Descriptor List:
>   "Serial Port" (0x1101)
>     Version: 0x0100

I installed pybluez using these commands:

> sudo apt-get install libbluetooth-dev
> sudo apt-get install python-dev
> sudo pip install PyBluez
> sudo pip3 install PyBluez

OS data

> DISTRIB_ID=Ubuntu
> DISTRIB_RELEASE=16.04
> DISTRIB_CODENAME=xenial
> DISTRIB_DESCRIPTION="Ubuntu 16.04.3 LTS"
> NAME="Ubuntu"
> VERSION="16.04.3 LTS (Xenial Xerus)"
> ID=ubuntu
> ID_LIKE=debian
> PRETTY_NAME="Ubuntu 16.04.3 LTS"
> VERSION_ID="16.04"

rgov, 2018-02-09:
First, please try installing the current version of PyBluez on GitHub rather than from PyPi. Let us know if it still occurs.

I don't have a system that matches your configuration. If you're able to figure it out, please report back or open a pull request so that others can benefit. Thanks.

matealex, 2018-02-12:
I tried installing the current version from GitHub, but i get the same results.

closed 2019-07-30 by stale[bot]