pybluez/pybluez #191: Descriptor in Bad State

faxe1008, 2018-03-02:
Hi,

I am trying to connect a HC-06 module using bluetooth, which is already paired with my device.
```
self.bluetooth_socket = bluetooth.BluetoothSocket(bluetooth.RFCOMM)
self.bluetooth_socket.connect(("MAC", 1))
```

and I am getting this error:
```
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (77, 'File descriptor in bad state')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "raspberrypi_code/BlueMQTTBroker.py", line 201, in <module>
    bluetooth_socket.connect((client.mac_address, client.port))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: (77, 'File descriptor in bad state')
```
I am really confused as to why that happens - a few weeks ago that worked just perfectly.

faxe1008, 2018-03-02:
It seems like this was caused by setting using `self.bluetooth_socket.settimeout(5)` removing it solves the problem however I need to set the timeout :^)

closed 2019-07-30 by stale[bot]

hoangesoftblog, 2020-08-02:
For those who looking for this problem, it is probably because the socket is already closed (by timeout, as someone mentioned above, or accidentally closed) when it connects to the address and the port. Create new socket helps solve that problem in my case.