pybluez/pybluez #193: Pull Request #95 broke travis-ci builds

btashton, 2018-03-12:
The build log from travis-ci shows the build breaking with:
```
gcc -pthread -fno-strict-aliasing -g -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I./port3 -I/opt/python/2.7.14/include/python2.7 -c bluez/btmodule.c -o build/temp.linux-x86_64-2.7/bluez/btmodule.o
bluez/btmodule.c: In function ‘init_bluetooth’:
bluez/btmodule.c:3688:22: error: ‘HCI_CHANNEL_USER’ undeclared (first use in this function)
     ADD_INT_CONST(m, HCI_CHANNEL_USER);
                      ^
bluez/btmodule.c:3069:60: note: in definition of macro ‘ADD_INT_CONST’
 #define ADD_INT_CONST(m, a) PyModule_AddIntConstant(m, #a, a)
                                                            ^
bluez/btmodule.c:3688:22: note: each undeclared identifier is reported only once for each function it appears in
     ADD_INT_CONST(m, HCI_CHANNEL_USER);
                      ^
bluez/btmodule.c:3069:60: note: in definition of macro ‘ADD_INT_CONST’
 #define ADD_INT_CONST(m, a) PyModule_AddIntConstant(m, #a, a)
                                                            ^
error: command 'gcc' failed with exit status 1
```

This is because the we are build with a really old version of bluez (4.101) which was released 6 years ago.
This pull request leverages the bluez User Channel that was added about 4 years ago:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/lib/hci.h?id=c0b7506c42f73a5ad2d4d05c14b2676e6a08edb9

If we can update the travis config to use at least `xenial` instead of `trusty` I think this will just work.

rgov, 2018-03-13:
Unfortunately we [can't](https://github.com/travis-ci/travis-ci/issues/5821). Can we `apt-get` the newer kernel?

btashton, 2018-03-15:
@rgov we don't actually need the new kernel to do the build, just the header files from the BlueZ library.  I just submitted a PR that should resolve this.  I did not see much point in actually installing the new version of the library (and it might require also updating the kernel which we cannot do) since we are not running tests against real bluetooth interfaces.

PR #195 (2018-03-15): Build in travis-ci against latest bluez-5.49 
This should resolve the issue where pybluez could not be built because it relies on new features that are not included in the old version of Bluez included in the version of Ubuntu Travis-CI uses.

Resolves #193
closed 2018-03-15 by rgov