pybluez/pybluez #196: Cannot bind RFCOMM socket

sourabhmarathe, 2018-04-02:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**
macOS 10.13.4
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
Macbook's bluetooth
 - **PyBluez Version:**
PyBluez==0.22

## Issue

I am trying to create a server socket that follows the example here: https://github.com/pybluez/pybluez/blob/master/examples/simple/rfcomm-server.py

When I call server_sock.bind(("",PORT_ANY)), I get the following error:
>>> server_sock.bind((target_address, 0))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/sourabh/pyblue/venv/lib/python2.7/site-packages/bluetooth/osx.py", line 110, in bind
    return self._sock.bind(addrport)
  File "/Users/sourabh/pyblue/venv/lib/python2.7/site-packages/lightblue/_bluetoothsockets.py", line 297, in bind
    address = (address[0], _getavailableport(self.__conn.proto))
  File "/Users/sourabh/pyblue/venv/lib/python2.7/site-packages/lightblue/_bluetoothsockets.py", line 62, in _getavailableport
    result, channelID, servicerecordhandle = BBServiceAdvertiser.addRFCOMMServiceDictionary_withName_UUID_channelID_serviceRecordHandle_(BBServiceAdvertiser.serialPortProfileDictionary(), "DummyService", None)
TypeError: Need 5 arguments, got 3

rgov, 2018-04-03:
Duplicate of #174. Please try installing PyBluez from the repo rather than PyPI and read that thread to see if there are other fixes that need to be made.

closed 2018-04-03 by rgov

tomasero, 2018-09-07:
Any news on a fix for this?