pybluez/pybluez #197: Miss bluetooth.py

kennyjudo, 2018-06-10:
I saw the example have to import "bluetooth.py", however, i can't find it in the file that i download from here, anyone can help? thanks! 

Kenny. 
 
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **PyBluez Version:**

## Issue

Please provide information necessary to reproduce the issue, including sample
code.

If you have long logs, please post them on https://gist.github.com and link to
the Gist here.

rgov, 2018-06-10:
You should review how to install Python packages using pip. Virtually all packages are installed this way, not by downloading individual files from GitHub.

closed 2018-06-10 by rgov

kennyjudo, 2018-06-17:
Hi Ryan,

got it, thanks for your remind, however, i still can install the pybluez
even i use the “python setup.py install” inside the file directory. i am
using win10 64bit anaconda and spyder3.2.8