pybluez/pybluez #209: ImportError : No module named ble

dreamtime1122, 2018-09-09:
I had install all the dependency and pybluez[ble] on my raspberry pi3 but still can't run the ble example.

> > pkg-config
> > libboost-python-dev
> > libboost-thread-dev
> > libbluetooth-dev >= 4.101
> > libglib2.0-dev
> > python-dev
> >
> gattlib
> 
> pybluez
> 
> pybluez[ble]
> 

```
from bluetooth.ble import DiscoveryService
service = DiscoveryService()
devices = service.discover(2)
for address, name in devices.items():
    print("name: {}, address: {}".format(name, address))
```

rgov, 2018-09-09:
This isn't an actionable bug report because you didn't include any error message.

rgov, 2018-09-10:
> Traceback (most recent call last):
> File "scan_test.py" , line 2 , in
> from bluetooth.ble import BeaconService
> ImportError : No module named ble

Can you run `pip uninstall 'pybluez[ble]'` and then `pip install -vvv 'pybluez[ble]'` and attach the output? (Please attach it as a file, or paste it into [Gist](https://gist.github.com), and do not just paste the whole thing into the comment box.)

closed 2019-07-30 by stale[bot]