pybluez/pybluez #211: BluetoothSocket has no 'type' and 'family' attributes

bbregeault, 2018-09-24:
BluetoothSocket is incompatible with asyncio's [connect_accepted_socket()](https://docs.python.org/3.6/library/asyncio-eventloop.html#asyncio.BaseEventLoop.connect_accepted_socket):
(Tested under linux with python3.6.6 and pybluez 0.22)
```python
Traceback (most recent call last):
  File "protocol_server.py", line 183, in connect_client
    transport, protocol = yield from self.loop.connect_accepted_socket(proto, client)
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1082, in connect_accepted_socket
    if not _is_stream_socket(sock.type):
AttributeError: 'BluetoothSocket' object has no attribute 'type'
```
and
```python
Traceback (most recent call last):
  File "protocol_server.py", line 183, in connect_client
    transport, protocol = yield from self.loop.connect_accepted_socket(proto, client)
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1087, in connect_accepted_socket
    sock, protocol_factory, ssl, '', server_side=True)
  File "/usr/lib/python3.6/asyncio/base_events.py", line 827, in _create_connection_transport
    transport = self._make_socket_transport(sock, protocol, waiter)
  File "/usr/lib/python3.6/asyncio/selector_events.py", line 73, in _make_socket_transport
    extra, server)
  File "/usr/lib/python3.6/asyncio/selector_events.py", line 694, in __init__
    _set_nodelay(self._sock)
  File "/usr/lib/python3.6/asyncio/selector_events.py", line 45, in _set_nodelay
    if (sock.family in {socket.AF_INET, socket.AF_INET6} and
AttributeError: 'BluetoothSocket' object has no attribute 'family'
```

A possible workaround is through monkey-patching:
```python
from socket import SOCK_STREAM, AF_BLUETOOTH
client.type = SOCK_STREAM
client.family = AF_BLUETOOTH
transport, protocol = yield from loop.connect_accepted_socket(asyncio.Protocol, client)
```

To reproduce the bugs:
```python
import asyncio
from socket import SOCK_STREAM, AF_BLUETOOTH
import bluetooth

@asyncio.coroutine
def connect_client(self, loop, sock):
    client, addr = yield from loop.sock_accept(sock)
#    client.type = SOCK_STREAM
#    client.family = AF_BLUETOOTH
    transport, protocol = yield from loop.connect_accepted_socket(asyncio.Protocol, client)

loop = asyncio.get_event_loop()
bt_sock = bluetooth.BluetoothSocket()
bt_sock.bind(("",bluetooth.PORT_ANY))
bt_sock.settimeout(0)
bt_sock.listen(1)
    
loop.run_until_complete(connect_client(loop, bt_sock))
```

It's probably a good idea to also add the 'proto' attribute, since these three are defined by cpython in the socket object : [source](https://github.com/python/cpython/blob/master/Modules/socketmodule.c#L4811)

rgov, 2018-09-24:
Thanks for filing this issue. No one is actively developing the project. If this is something you would like to contribute, and can maintain backwards compatibility with Python 2.7, I can merge it in for you.

bbregeault, 2018-09-24:
Alright,
I'm not a professional developer so I don't have much time at the moment but I'll give it a try when my timetable clears up.

rgov, 2018-09-24:
Thank you! No need to be a professional developer, most open source projects are developed by hobbyists. Whenever you have time, your contribution is appreciated.

closed 2019-07-28 by bbregeault