pybluez/pybluez #212: pip has an old version of pybluez published (doesn't work on OSX)

vlovich, 2018-09-27:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** OSX 10.13
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.) N/A
 - **PyBluez Version:** 0.22

## Issue

pip install pybluez fails on OSX.  Seems like the version published is really old (0.22) & likely doesn't have the version master does.

https://gist.github.com/vlovich/12c9eff5aeb1eeecc36cb2513a5f4685

rhubarbdog, 2018-10-19:
i think i'm falling foul of something similar, i'm running raspbery pi rasbian 9.2 and installed pybluez with command `sudo -H pip3 install pybluez`. when i try to run the code rfcomm-server.py from your README.md i get the following error.
```
$ python3 rfcomm-server.py 
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/bluetooth/bluez.py", line 240, in advertise_service
    protocols)
_bluetooth.error: (2, 'No such file or directory')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "rfcomm-server.py", line 20, in <module>
    profiles = [ SERIAL_PORT_PROFILE ], 
  File "/usr/local/lib/python3.5/dist-packages/bluetooth/bluez.py", line 242, in advertise_service
    raise BluetoothError (str (e))
bluetooth.btcommon.BluetoothError: (2, 'No such file or directory')
```

rgov, 2019-07-23:
I'm hoping to get to this soon.

Arusekk, 2019-12-27:
Bump. In several days Python 2 will no longer be supported, so if compatibility was the problem, it is not an issue now.

rgov, 2019-12-27:
The biggest blocker is building for Windows. It seems like there is some difficulty building for Windows 10. Additionally it is recommended to install multiple versions of Visual Studio for building extensions for different every minor release of Python, which is time consuming.

I could just ask someone else to do it and send the binary files, but that makes me uncomfortable because I cannot verify the source has not been modified.

One thing I can do is try to use GitHub Actions to build for Windows, but I haven't yet been successful. If you have the ability to contribute support for that, it would be very helpful.

Note there are no developers actively working on this project and I am just a volunteer who keeps an eye on pull requests to fix broken code.

Arusekk, 2019-12-27:
I'm just a contributor, so I cannot modify Actions, but the interface for building on all platforms should be straightforward enough.

Also why not create some beta-release, or source-only release for now, with wheels/binaries added later? It would be just an easier equivalent of doing `pip install git+https://.....` stuff. My point is that a newer release can always be uploaded some day if it happens to be ready.

rgov, 2019-12-27:
> I'm just a contributor, so I cannot modify Actions

You can fork the repo and set up actions on your fork, then suggest a pull request. I have some progress on this at rgov/pybluez.

> Also why not create some beta-release, or source-only release for now

Yes that probably could be done, I'll look into it.

rgov, 2019-12-28:
PyBluez 0.23 has been released. Please try installing it on macOS and let me know if it works.

closed 2019-12-28 by rgov