pybluez/pybluez #215: pybluez broken on OSX w/ python3

vlovich, 2018-09-28:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** 10.13.6
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **PyBluez Version:** HEAD

## Issue

Try invoking send("abc").  Throws an exception at the OSX layer because "abc" is not a buffer object.
```
732     def write(self, data):                                                      
733         if self.channel is None:                                                
734             raise _socket.error("socket not connected")                         
735         return \                                                                
736             BBBluetoothChannelDelegate.synchronouslyWriteData_toRFCOMMChannel_( 
737                 Foundation.NSData.alloc().initWithBytes_length_(data, len(data)),
738                 self.channel)                                                   
```

initWithBytes needs a byte-like object which string is not in python3.

rgov, 2018-09-28:
Thanks. Can you suggest a fix in a pull request?

vlovich, 2018-09-28:
I actually don't have an idea of exactly what the right fix is.  Should send/recv expect bytes instead of string?  I think so but that would impact existing users.  Should we call encode("utf-8")?  We could but that's kind of not great in case you want to send binary data which may not encode to UTF-8.  The cleanest is probably to utf-8 encode internally at the entrypoint & pass around bytes everywhere.  However, that would require changing Linux & OSX and I don't have environments setup to test that.  Any chance you can contribute those parts?

rgov, 2018-09-28:
Unfortunately I am not developing much here, I just stepped in to try to help merge in pull requests. I agree that bytes should be used most places. I suggest doing UTF-8 encoding only if running on Python 2.7, otherwise the client code should be updated to send the correct type. (A helper function might be useful for this.)

You are developing on macOS, right? Is it only Linux that you would need help testing on?

vlovich, 2018-09-29:
Presumably Windows too.

PR #216 (2018-10-01): Fix OSX send compatability with python3
We actually have to use bytes not strings. In Python3 these are distinct types whereas in Python2
bytes is an alias for strings.

Fixes #215
closed 2018-10-01 by rgov