pybluez/pybluez #224: Irprops is not available on newer version of Windows 10; consider changing to Bthprops

irvinec, 2018-10-25:
## System

 - **Operating System:** Widnows 10 1809 (OS Build 18262.100)
 - **Hardware:** PC with external USB Bluetooth adapter.
 - **Python Version:** Python 3.7.1
 - **PyBluez Version:** 0.22

## Issue

In setup.py currently, _msbt.c depends on irprops:
```python
ext_modules.append(Extension('bluetooth._msbt',
                          include_dirs=["%s\\Include" % PSDK_PATH, ".\\port3"],
                          library_dirs=[lib_path],
                          libraries=["WS2_32", "Irprops"], # <-- here is the irprops reference
                          sources=['msbt\\_msbt.c']))
```

At least on the build of Windows 10 I'm running, irprops is not available (not in \Windows\System32).
This causes a DLL load error when trying to `import bluetooth._msbt as bt` in msbt.py:

Traceback (most recent call last):
  File "somefile.py", line 8, in <module>
    from bluetooth_interface import BluetoothInterface
  File "somefile.py", line 4, in <module>
    import bluetooth # pybluez
  File "C:\Users\user\AppData\Local\Programs\Python\Python37\lib\site-packages\pybluez-0.22-py3.7-win-amd64.egg\bluetooth\__init__.py", line 39, in <module>
  File "C:\Users\user\AppData\Local\Programs\Python\Python37\lib\site-packages\pybluez-0.22-py3.7-win-amd64.egg\bluetooth\msbt.py", line 2, in <module>
  File "C:\Users\user\AppData\Local\Programs\Python\Python37\lib\site-packages\pybluez-0.22-py3.7-win-amd64.egg\bluetooth\_msbt.py", line 7, in <module>
  File "C:\Users\user\AppData\Local\Programs\Python\Python37\lib\site-packages\pybluez-0.22-py3.7-win-amd64.egg\bluetooth\_msbt.py", line 6, in __bootstrap__
  File "C:\Users\user\AppData\Local\Programs\Python\Python37\lib\imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: DLL load failed: The specified module could not be found.

It looks like irprops exposed bluetooth functionality at one point, but that functionality has been moved over to bthprops.
See MSDN documentation for one of the functions _msbt.c is referencing: [BluetoothFindFirstDevice](https://docs.microsoft.com/en-us/windows/desktop/api/bluetoothapis/nf-bluetoothapis-bluetoothfindfirstdevice)

I made this change locally and I am now able to install and import pybluez without error.
```python
ext_modules.append(Extension('bluetooth._msbt',
                          include_dirs=["%s\\Include" % PSDK_PATH, ".\\port3"],
                          library_dirs=[lib_path],
                          libraries=["WS2_32", "Bthprops"], # <-- changed irprops to bthprops
                          sources=['msbt\\_msbt.c']))
```

## Notes

I diagnosed the issue with [Dependencies](https://github.com/lucasg/Dependencies).
Examining the .pyd file for _msbt I saw that irprops was a missing dependency.
After looking up what functions irprops normally exposes, I found that bthprops also exposed those functions, was recommended by MSDN, and was available on my system.

I'm happy to submit a PR for the change, but I'm not sure how it might effect older versions of Windows or even when irprops was removed.

closed 2019-07-30 by stale[bot]

SU-quinten, 2021-01-07:
I would like to add that I had the same issue and the solution provided by irvinec solved my problems. I would suggest either a mention on the homepage or submit this change.

rgov, 2021-01-07:
The change has already been merged in, I think. Did you try from the latest sources in git?

robinkanters, 2021-01-10:
Installing from sources fixed the issue for me

(`pip install git+https://github.com/pybluez/pybluez@master`)

commit d7f36702 (2020-07-14): Fix #224: Dll load error when importing on Windows 10... (#226)

Changed Irprops dependency for _msbt to Bthprops.
Bthprops is the supported and recommended bluetooth library on Windows.
Irprops is only present for driver backward compatibility.
Irprops may not be available on newer versions of Windows.