pybluez/pybluez #230: Set scan parameter failed : Are you root on raspberry pi3

wan420, 2018-12-26:
I am running ble scan code from the default example given here, but all the time it gets stuck at "scan parameter failed: are you root?" error

I am using python3 for this and its a Raspian noobs OS in raspberry pi 3

Even if I run this with sudo command in the terminal I still get the same error.
Any advice as to what am i doing wrong..
here's the code : 
 ```
from bluetooth.ble import DiscoveryService
service = DiscoveryService()
devices = service.discover(2)
for lines in list(devices.items()):
    print(lines[1])
```

and here's the trace : 
```
Traceback (most recent call last):
  File "/home/pi/raspible/bletrial.py", line 3, in <module>
    devices = service.discover(2)
RuntimeError: Set scan parameters failed (are you root?)
```
Thanks

H-Kz, 2019-01-07:
I want you to try it in ROOT user or "sudo python..."

wan420, 2019-01-10:
Even if I do that , I was still getting that problem, So i reinstalled the raspi and reinstalled the libraries, the problem hasnt occured yet. I will close this issue, will reopen if the problem reappears. 
Thans

closed 2019-01-10 by wan420

raj-priyanshu, 2019-12-09:
Hi,
I was trying to run your example
`
# bluetooth low energy scan
from bluetooth.ble import DiscoveryService

service = DiscoveryService()
devices = service.discover(2)

for address, name in devices.items():
    print("name: {}, address: {}".format(name, address))
`
but it keeps giving the r u root error again and again. I tried to run as root also used sudo but the result is still the same. I tried this on raspberry pi and ubuntu-desktop both. please look into this.

`
>>> devices = service.discover(2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Set scan parameters failed (are you root?)
`

aprilahijriyan, 2020-03-08:
I also experienced the same thing

MAKER-ACADEMY, 2021-02-17:
Hello,I have same problem with python 3 raspberry pi ,and Dongle BLE 4.0 , i
"run time error set scan parameters failed (are you root ) "
Help please !!
code:

from gattlib import DiscoveryService

service = DiscoveryService("hci0")
devices = service.discover(2)

for address, name in devices.items():
    print("name: {}, address: {}".format(name, address))

zhikuan-aw, 2021-03-28:
Anyone??

rajithavk, 2021-05-29:
I have the same problem on a beaglebone running bluez 5.58 and the hardware module is a nordic52840 with the latest stack.
Running btmon gives the following output.

< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                                                                                                                                                                                           #1 [hci0] 5.137842
        Type: Active (0x01)
        Interval: 10.000 msec (0x0010)
        Window: 10.000 msec (0x0010)
        Own address type: Public (0x00)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                          #2 [hci0] 5.139957
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Command Disallowed (0x0c)

So I think this is an issue with the hci parameter misconfiguartion internally at gattlib.

steveswinbank, 2022-01-27:
Doing this worked for me:
```
sudo hciconfig hci0 down
sudo hciconfig hci0 up
```

Then:
```
sudo service bluetooth restart
```